realsenseai/librealsense #1856: Alignment of captured images from Realsense viewer

adarvit, 2018-06-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400 | 
| Firmware Version                   | | 
| Operating System & Version |   Win10  | 
| Platform                                 | PC  |
| SDK Version                            |   2.11.1                         | 
| Language                            | python                    | 

I have some images (rgb and depth raw) i captured using the snapshot button of realsense viewer
I tried to align the depth image to the rgb image using the following code:
`            depth_point = d435.processing.rs2_deproject_pixel_to_point(depth_Intrinsics, depth_pixel,depth_value*d435.processing.depth_scale)
`
`            color_point = d435.processing.rs2_transform_point_to_point(d435.processing.depth_extrinsics['848'], depth_point)
`
`            color_pixel = d435.processing.rs2_project_point_to_pixel(d435.processing.color_Intrinsics['848'], color_point)
`
but the images were not aligned
I used the depth_intrsincs that i got from the depth_metadata when i captured the depth frame, which is different from the one i get when i am using the api (for the color Intrinsic and depth to color extrinsic i am using the api).

How can i align images i captured from real sense viewer snapshot button?

RealSense-Customer-Engineering, 2018-06-19:
[Realsense Customer Engineering Team Comment]
@adarvit
the rs2_deproject_pixel_to_point function in rsutil.h accepts the parameters the order should be `static void rs2_deproject_pixel_to_point(float point[3], const struct rs2_intrinsics * intrin, const float pixel[2], float depth)`​. But from your code your parameter obviously not the same, I don't think python wrapper changes the parameter order, can you attach your whole python code? Also for the mapping, you can refer to the **align.cpp** function "**align_z_to_other**" which illustrate how to map the depth image to color.

RealSense-Customer-Engineering, 2018-07-06:
[Realsense Customer Engineering Team Comment]
@adarvit
Any more issue?

closed 2018-07-12 by RealSense-Customer-Engineering

Resays, 2019-06-20:
can you share the code ?
i am working something exactly same so i need to view your code ... can i have it ?