realsenseai/librealsense #1857: Unity Ros Reader

frenchtouchXR, 2018-06-08:
Hello, 

I tried to enable the recorder and reader on the Unity Wrapper. The recorder works well but when I try to read the recorded file I have this error : 

Exception: Failed to resolve request. Request to enable_device_from_file("my_file") was invalid, Reason: Failed to create ros reader: Bag unindexed
rs2_config_resolve(config:000000003A92A230, pipe:000000003A929D90) 

What do you think about that ? 

Best regards

François

dorodnic, 2018-06-08:
Hi @frenchtouchXR 
Usually, `Bag unindexed` error indicates that the recording was not properly closed. When using the .NET wrapper it is especially critical to make sure all wrapper objects are cleaned-up using `.Dispose()` before closing the application / game.

frenchtouchXR, 2018-06-08:
Thanks for your quick answer. You're talking about the .net Dispose() fonction. I ll try it soon even if I work in Unity in c#. More over I tried To read in Unity a .bag file recorded in the intelRSviewer I Will get the "failed to create ros reader error

frenchtouchXR, 2018-06-14:
Any news so far ?

RealSense-Customer-Engineering, 2018-06-28:
[Realsense Customer Engineering Team Comment]
Hi @frenchtouchXR,

Need to check internally about this feature request and see how soon it can be available.

RealSense-Customer-Engineering, 2018-07-26:
[Realsense Customer Engineering Team Comment]
Hi @frenchtouchXR,

Please try librealsense 2.14.0 to see how it works.

frenchtouchXR, 2018-07-31:
Hello,

It works fine. Thank you. I' m now trying to make a multicam rec on Unity. I changed the static function in the realsensedevice inspector script to a public one. Then I try to get the realsense device by entering its serial number from the other scripts. I get this error : 

The same field name is serialized multiple times in the class or its parent class. This is not supported: Base(MonoBehaviour) myRealSenseDevice

frenchtouchXR, 2018-07-31:
I think i il close this thread and open a new one.

closed 2018-07-31 by frenchtouchXR