realsenseai/librealsense #1858: Multicam not possible in Unity wrapper through RealSenseDevice.cs script ?

iQuentin, 2018-06-08:
---------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435                                                       | 
| Firmware Version                   | 05.09.11.00                                             | 
| Operating System & Version |  Win 10                                                   | 
| Platform                                 | PC                                                          |
| SDK Version                            |   2.11.1                                                   | 
| Language                               |  C/C#/unity                                              | 

### Issue Description

Hi community,
i'm trying to modify the Unity wrapper project (scene : _colored point cloud.unity_) to show 2 or more point clouds from 2 or more depth cams :
The _RealSenseDevice.cs_ script has a **RequestSerialNumber** field which seems to be made for that, and indeed, when i change this serial i can choose another cam. BUT, when i duplicate the _RealsenseDevice_ GameObject in the editor to get two different cams with different serials, the same cam is used to generate two point clouds (please note that i had to tweak the script a little to make this work because it doesn't allow multiple instances in its original shape, but this is not the subject). And in fact, we can see in the editor in the _RealSense Device Inspector_ that the same serial number is found by the two GameObjects.
After investigations in the code, i see that this function is called : _Config.EnableDevice(RequestedSerialNumber)_ which wraps this realsense c++ function : _rs2_config_enable_device_. Furthermore, in the header comments of this function we can read 

> Select a specific device explicitly by its serial number, to be used by the pipeline...

  (see [here](https://github.com/IntelRealSense/librealsense/blob/v2.8.0/include/librealsense2/h/rs_pipeline.h#L231) )

So i'm confused...
Does it speak to someone?

dorodnic, 2018-06-08:
Hi @iQuentin 
I think it should be working, we will look into it.

iQuentin, 2018-06-12:
Oops my mistake. Further investigations told me that my little and quick tweak about the instances was too... quick. The scripts are shaped for a single instance of RealSenseDevice and i artificially made 2 instances but the other scripts silently took only the first one as reference. That explains everything, and it's working as expected now. This was a non-subject, my apologies.

closed 2018-06-12 by iQuentin

YannCubicMovie, 2018-07-25:
Hi @iQuentin!
I had the same issue as you and I also figured out the single RealSenseDevice instance issue.

I removed the RealSenseDevice static property and added a way to select the right RealSenseDevice  instance in the related classes (in RealSenseStreamTexture, RealSenseDeviceInspector among others)
The result is that I can display my 2 different pointcloud with different depths coming from each of the two cameras I use BUT the color texture of the two point clouds is the same (coming from the same camera)...

Did you get the same issue? If yes, did you find a workaround?

iQuentin, 2018-07-26:
Hi Yann,
I did not get the same issue, so i am sorry i can't help you.
Anyway i understood that the different camera model (mine is D435) can have different ways of handling each of their data channels... if it can help.

akashcastelino, 2019-01-28:
Hi @YannCubicMovie 

Were you able to solve the issue with the same color texture being applied to the pointclouds of all cameras ?
Facing the same issue here.

jian-chai, 2019-06-27:
> Hi @iQuentin!
> I had the same issue as you and I also figured out the single RealSenseDevice instance issue.
>
> I removed the RealSenseDevice static property and added a way to select the right [...]

> static

HI

> Hi @iQuentin!
> I had the same issue as you and I also figured out the single RealSenseDevice instance issue.
>
> I removed the RealSenseDevice static property and added a way to select the right [...]

> Hi @iQuentin!
> I had the same issue as you and I also figured out the single RealSenseDevice instance issue.
>
> I removed the RealSenseDevice static property and added a way to select the right [...]

Hi Yann,
how did you work around this issue?