realsenseai/librealsense #1859: Accesss to frame data

ToshiEAB, 2018-06-09:
Camera mode: D415
Operating System & Version: Windows 10
Platform: PC
SDK Version: 2.10.3
Language: C++

### Issue Description
I define a depth frame as follows:

rs2:pipeline pipe;
rs2::config cfg;
rs2::frameset frames;
rs2::frame depthframe;

cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 60);
pipe.start(cfg);

frames = pipe.wait_for_frames();
depthframe = frames.get_depth_frame();

Now, I'd like to modify the data of depth frame.  How can I access to the data at each pixel? I am aware of that depthframe.get_data() gives me a pointer to the start of the frame data. But I don't know how to access to the data at a particular pixel. Specifically, I'd like to do something equivalent to the "transfer_pixel()" function in align.cpp.

Regards,

dorodnic, 2018-06-09:
Hi @ToshiEAB 
Something like this:
```cpp
auto w = depthframe.get_width();
auto pixels = (uint16_t*)depthframe.get_data();
auto p = pixels[y * w + x];
```

ToshiEAB, 2018-06-10:
Thanks dorodnic. It works. 

Now I have additional question similar to this one: https://github.com/IntelRealSense/librealsense/issues/956

Is there now any way to create a "blank" frame (in the form of rs2::frame)? Following the code you showed me above, I want to copy the value at each pixel of depth frame to the corresponding pixel on an aligned depth frame, just like the "transfer_pixel(...)" function in align.cpp.

I tried creating it, using memcpy, memset, etc., but it's been unsuccessful.

Thanks,

RealSense-Customer-Engineering, 2018-06-11:
[Realsense Customer Engineering Team Comment]
@ToshiEAB
you can reference the code snippet below and also the rs-measure.cpp to achieve what you want.

    processing_block pb([&](rs2::frame f, const rs2::frame_source& source)
    {
     // source can be used to allocate new frames and send them out source.frame_ready(<frame_created>);
    });
    frame_queue q;
    pb.start(q); // Results from the block will be enqueued into q
    pb.invoke(input_frame); // Invoke the lambda above
    frame output_frame = q.wait_for_frame(); // Fetch the result, result will be from above "source.frame_ready" function.

From the rs-measure.cpp, you can refer to the block 188 for the whole picture how to use the processing_block:
**video_process_thread** thread creates **frame_processor** processing block, and the line "`frame_processor >> postprocessed_frames`" equals to "`frame_processor.start(postprocessed_frames)`". At the bottom part of the thread lambda function, the while loop keep retrieving the frame from pipe then feed into frame_process.invoke. The process then goes into the **processing_block segment(frame_processor lambda function)**, then you see the last line inside the segment "source.frame_ready(combined)" which will push the combined frame into the **postprocessed_frames** queue for other process to retrieve.

ToshiEAB, 2018-06-12:
@dorodnic Thanks for the information. I figured out how the code snippet you mentioned works. But what should I write inside the lambda function (other than source.frame_ready(<frame_created>)) to create a new blank frame in the rs2::frame format, which does not share any memory with the input frame? (A deep copy of the input frame also is acceptable to me.)

Thanks

RealSense-Customer-Engineering, 2018-06-12:
[Realsense Customer Engineering Team Comment]
@ToshiEAB
Please refer to similar issue which contains what you need.

https://github.com/IntelRealSense/librealsense/issues/1374

ToshiEAB, 2018-06-12:
OK, I figured it out! Thanks all for your support!

closed 2018-06-12 by ToshiEAB