realsenseai/librealsense #1860: Video4linux gets removed on every reboot in /sys/class

SteveMacenski, 2018-06-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (05.09.09.02) | 
| Operating System & Version |   {Ubuntu 16 | 
| Kernel Version (Linux Only)    |  (e.g. 4.13.0-43-generic)                                         | 
| Platform                                 | Intel Nuc  |
| SDK Version                            |  { l2.10.2 }                          | 
| Language                            |  {C }                          | 

### Issue Description
Every time I reboot the machine the /sys/class/Video4Linux work goes away and the cameras cannot connect. lsusb shows the camera connected. If I unplug and plug them back in, they'll start working, but never on boot up. Subsequent restarts of the program work find, but reboots do not.  

I see this error 
` 10/06 00:34:41,459 ERROR [140364363966336] (backend-v4l2.cpp:402) Cannot access /sys/class/video4linux`

from the ROS driver

ev-mp, 2018-06-11:
@SteveMacenski , 
The failure to load the kernel module could be due to the PC power management that interferes with the devices enumeration process. It also could be due to host-related/installation issue.

First, please verify that you have the  xHCI mode in BIOS turned on, as described in #1283.
Then inspect the enumeration process by looking into the kernel log after reboot and look for USB errors such as mentioned in #1615.  You can share the kernel log here as well.

In case the device is enumerated properly, but the kernel module is still not loaded during boot-up by Ubuntu you may enforce that by adding uvc driver to the startup list
`echo 'uvcvideo' | sudo tee -a /etc/modules`.
though I would recommend it only as the last resort.

More questions:
- Is this behavior persistent when switching USB ports on the NUC?
- Kernel 4.13.0-43 is not yet available by `sudo apt dist-upgrade` - did the PC work correctly with other (older)  kernel versions?

SteveMacenski, 2018-06-11:
The behavior is persistent with USB port changes, even when changing to ports that are always powered. I have just bootstrapped the machine so it hasnt been running any other kernal. 

I found that partial issues were due to the Video4Linux install issue requiring me to manually copy a script into a correct directory and rerunning, and the cable I was using apparently wasn't high quality. 

Will test your suggestions today.

SteveMacenski, 2018-06-12:
I'm using a 7th gen i5 NUC, I believe it only has xHCI mode from those tickets.

I'm still having the issue. When I run `sudo modprobe uvcvideo` then launch the ROS files, I see that the error ` Cannot access /sys/class/video4linux` goes away but still doesn't identify the cameras that are plugged in. lsusb still shows them.

Syslog shows the following

```Jun 11 18:23:43 walle kernel: [   34.885821] usb 1-2: new high-speed USB device number 5 using xhci_hcd
Jun 11 18:23:43 walle kernel: [   35.032399] usb 1-2: New USB device found, idVendor=8086, idProduct=0b07
Jun 11 18:23:43 walle kernel: [   35.032401] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Jun 11 18:23:43 walle kernel: [   35.032402] usb 1-2: Product: Intel(R) RealSense(TM) 435
Jun 11 18:23:43 walle kernel: [   35.032403] usb 1-2: Manufacturer: Intel(R) RealSense(TM) 435
Jun 11 18:24:18 walle kernel: [    1.732033] usb 1-2: new high-speed USB device number 3 using xhci_hcd
Jun 11 18:24:18 walle kernel: [    1.872756] usb 1-2: New USB device found, idVendor=8086, idProduct=0b07
Jun 11 18:24:18 walle kernel: [    1.872760] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Jun 11 18:24:18 walle kernel: [    1.872764] usb 1-2: Product: Intel(R) RealSense(TM) 435
Jun 11 18:24:18 walle kernel: [    1.872767] usb 1-2: Manufacturer: Intel(R) RealSense(TM) 435
Jun 11 18:24:18 walle kernel: [    1.873294] usb 1-2: can't set config #1, error -32
Jun 11 18:27:33 walle kernel: [  197.974950] usb 1-2: USB disconnect, device number 3
Jun 11 18:27:41 walle kernel: [  206.277470] usb 1-2: new high-speed USB device number 6 using xhci_hcd
Jun 11 18:27:41 walle kernel: [  206.422815] usb 1-2: New USB device found, idVendor=8086, idProduct=0b07
Jun 11 18:27:41 walle kernel: [  206.422822] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Jun 11 18:27:41 walle kernel: [  206.422827] usb 1-2: Product: Intel(R) RealSense(TM) 435
Jun 11 18:27:41 walle kernel: [  206.422831] usb 1-2: Manufacturer: Intel(R) RealSense(TM) 435
Jun 11 18:28:21 walle kernel: [    1.740031] usb 1-2: new high-speed USB device number 3 using xhci_hcd
Jun 11 18:28:21 walle kernel: [    1.880718] usb 1-2: New USB device found, idVendor=8086, idProduct=0b07
Jun 11 18:28:21 walle kernel: [    1.880722] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Jun 11 18:28:21 walle kernel: [    1.880726] usb 1-2: Product: Intel(R) RealSense(TM) 435
Jun 11 18:28:21 walle kernel: [    1.880729] usb 1-2: Manufacturer: Intel(R) RealSense(TM) 435
Jun 11 18:28:21 walle kernel: [    1.881259] usb 1-2: can't set config #1, error -32
```

RealSense-Customer-Engineering, 2018-06-19:
[Realsense Customer Engineering Team Comment]
Hi @SteveMacenski,

Per your log, although D435 is connected to a XHCI port, it still enumerated as a high-speed USB device, instead of super-speed USB3 device. Suggest to upgrade your firmware to the latest version (ex: 5.9.11) and change a USB3 qualified cable.
------------------------------------
Jun 11 18:27:41 walle kernel: [206.277470] usb 1-2: new high-speed USB device number 6 using xhci_hcd
 Jun 11 18:27:41 walle kernel: [206.422815] usb 1-2: New USB device found, idVendor=8086, idProduct=0b07
------------------------------------

SteveMacenski, 2018-06-19:
I have done both of those things. I finally resolved by swapping the cable and manually editing a bunch of the install scripts.

closed 2018-06-19 by SteveMacenski