realsenseai/librealsense #1861: CUDA with RealSense SDK2.0

ToshiEAB, 2018-06-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   |  | 
| Operating System & Version |   Win 10 | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC  |
| SDK Version                            |  2.10.3 }                          | 
| Language                            |  C++                          | 

### Issue Description
I'm trying to replace align.cpp and align.h with align.cu and align.cuh, respectively for using CUDA. I have not made any change in their source codes -- just the name of files. I could activate the CUDA option on CMake. When building realsense2.dll on Visual Studio 2015, however, it gives me errors such as the following ones:

C2039   'HostingClass': "global namespace" is not a member .
C2433   'librealsense:HostingClass': 'friend' cannot be use to declare data.

The problem seem either that the align.cu and align.cuh are not compiled correctly or that these files are not linked to other C++ files correctly. Can anyone guess what changes I need to make?

P.S. Because I changed the file name, I used Find in Files on Visual Studio and changed the file name in all relevant files. But this did not solve the problem above.

Regards,

dorodnic, 2018-06-11:
Hi @ToshiEAB 
I don't fully understand your approach, but we are working on adding out-of-the-box CUDA support and planning to publish it soon

ToshiEAB, 2018-06-11:
@dorodnic My apologies for not being very clear. But I am glad to hear that CUDA support will be added to the SDK soon.

RealSense-Customer-Engineering, 2018-06-14:
[Realsense Customer Engineering Team Comment]
@ToshiEAB
Per dorodnic said, the CUDA out-of-the-box solution should be published soon, please keep tracking LibRS repository and test it once released, if any further question, welcome to file new issue. I close this ticket first.

closed 2018-06-14 by RealSense-Customer-Engineering