realsenseai/librealsense #1863: How to manually patch the kernel for INZI?

KeCh96, 2018-06-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Firmware Version                   |  | 
| Operating System & Version |   Debian  | 
| Kernel Version (Linux Only)    |  4.11.0 | 
| Platform                                 | ARM board |
| SDK Version                            |  SDK2  | 

I want to get depth frames and process by ARM board with Debian system. I build `pyrealsense2.so` in my ARM board. Then I run the following code in python:

> import pyrealsense2 as rs
> pipeline = rs.pipeline()
> pipeline.start()

I run into an error: 
> RuntimeError: The requested pixel format 'INZI' is not natively supported by the Linux

According to [dorodnic's answer](https://github.com/IntelRealSense/librealsense/issues/1785), I can manually patch the kernel or run cmake with `-DFORCE_LIBUVC=true`.

I run cmake with `-DFORCE_LIBUVC=true` but get `Segmentation fault` after several frames. I asked [a question](https://github.com/IntelRealSense/librealsense/issues/1821) but nobody give solution.

So it seems that run cmake with `-DFORCE_LIBUVC=true` not work for me; therefore, the only solution is manually patch the kernel. I try to follow [your answer](https://github.com/IntelRealSense/librealsense/issues/1186) @dorodnic : 
```
You will need to add couple lines of code to /drivers/media/usb/uvc/uvc_driver.c, /drivers/media/usb/uvc/uvcvideo.h and /include/uapi/linux/videodev2.h to make V4L2 recognise INZI. You can use the patch for Ubuntu as reference.
Once you make the changes and successfully rebuild the kernel, it would be relatively easy to generate new .patch that you are welcomed to share here for future use.
```

But I do not find `/drivers/media/usb/uvc/uvc_driver.c` or `uvcvideo.h` or `videodev2.h` in my Debian system. There is no `/driver` directory in my system. Do I need to download something? @dorodnic

dorodnic, 2018-06-11:
Hi @KeCh96 

Crash with libuvc seem to be exclusive to ARM boards, and we need time to reproduce and resolve it. 

To manually patch the kernel the high level flow is:
1. Download the sources of the kernel you are using (line 53 in the script bellow) 
2. Navigate to the relevant drivers folder and apply the change (line 88)
3. Build the kernel module (you don't have to build the whole kernel, just run `make` in the module directory) - line 125
4. Replace your existing module with newly built one using `modprobe` (line 141)

For more details you can follow one of the maintained scripts:
[patch-realsense-ubuntu-lts.sh](https://github.com/IntelRealSense/librealsense/blob/master/scripts/patch-realsense-ubuntu-lts.sh)

dorodnic, 2018-06-11:
Also worth mention that if you have an option to upgrade to kernel 4.12, this will also solve the problem (since SR300 support was upstreamed as part of this version)

KeCh96, 2018-06-11:
I typed `aptitude search linux-image | awk '{print $2}'` and get:

```
linux-image-4.11.0-qcomlt
linux-image-4.11.0-qcomlt-arm:armhf
linux-image-4.11.0-qcomlt-arm64
linux-image-4.11.0-qcomlt-arm64-dbg
linux-image-4.11.0-qcomlt-dbg
linux-image-4.14.0-qcomlt-arm:armhf
linux-image-4.14.0-qcomlt-arm64
linux-image-4.14.0-qcomlt-arm64-dbg
linux-image-4.9.0-6-arm64
linux-image-4.9.0-6-arm64-dbg
linux-image-4.9.0-qcomlt
linux-image-4.9.0-qcomlt-dbg
linux-image-arm64
linux-image-arm64-dbg
```

Does it mean my ARM board do not support kernel 4.12 ? @dorodnic

KeCh96, 2018-06-11:
I tried to run `./scripts/patch-realsense-ubuntu-lts.sh`, and get error: 
```
Remove all RealSense cameras attached. Hit any key when ready                                                                                
                                                                                                                                             
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package linux-headers-generic is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'linux-headers-generic' has no installation candidate

```

Then I run `apt-get update`, and get the following error (this error seems to be caused by `http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/xenial/InRelease` 403 Forbidden)
```
Ign:1 http://mirrors.163.com/debian stable InRelease                                                                                         
Hit:2 http://mirrors.163.com/debian stable Release                                                                                           
Err:4 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial InRelease                                                           
  403  Forbidden [IP: 52.218.16.129 80]                                                                                                      
Ign:5 http://obs.linaro.org/linaro-overlay-buster/buster ./ InRelease                                                                        
Ign:6 http://obs.linaro.org/qcom/buster ./ InRelease                                                                                         
Hit:7 http://obs.linaro.org/linaro-overlay-buster/buster ./ Release                                                                          
Hit:9 http://obs.linaro.org/qcom/buster ./ Release                                                                                           
Reading package lists... Done                                                                                                                
E: Failed to fetch http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/xenial/InRelease  403  Forbidden [IP: 52.218.16.129 80] 
E: The repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial InRelease' is not signed.                              
N: Updating from such a repository can't be done securely, and is therefore disabled by default.                                             
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

dorodnic, 2018-06-11:
Hi @KeCh96 

1. 4.14 version will also work
2. `./scripts/patch-realsense-ubuntu-lts.sh` is for x86/x64 Ubuntu and will not run out of the box on Arm, but you can see what it is doing for reference
3. It seems you have added `http://realsense-hw-public.s3.amazonaws.com/` to your apt sources, please remove it and try again

RealSense-Customer-Engineering, 2018-06-19:
[Realsense Customer Engineering Team Comment]
Hi @KeCh96,

Do you have any update according to @dorodnic's comments?

RealSense-Customer-Engineering, 2018-06-20:
[Realsense Customer Engineering Team Comment]
Hi @KeCh96,

I verified SR300 stream on one Jetson TX2 (ARM) board with librealsense 2.12.0 + FW 5.9.11. Either libuvc or kernel patch, both can work normally with the example "python-tutorial-1-depth.py". (more than 10 minutes)
What's your ARM board?

RealSense-Customer-Engineering, 2018-07-09:
[Realsense Customer Engineering Team Comment]
Hi @KeCh96,

Is the kernel patch ok on your ARM platform? Does "Segmentation fault" still happen when SR300 depth stream?
What's your SR300 firmware version? 3.21.0.0?

RealSense-Customer-Engineering, 2018-07-16:
[Realsense Customer Engineering Team Comment]
Hi @KeCh96,

any update?