realsenseai/librealsense #1867: RealSense Viewer not starting / no video rendering in SDK

Julz-s, 2018-06-11:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 }                                                   | 
| Firmware Version                   | (Open RealSense Viewer --> Should be 5.9.11 | 
| Operating System & Version | Windows 10       |                                    | 
| Platform                                 | PC                                                           |
| SDK Version                            | SDK 2.0 (Build 2.12.0)                            | 

### Issue Description
I cannot launch the RealSense viewer nor any other application which renders a video.
To be more specific: When starting the RealSense viewer, only a white window appears. Similary, I cannot see any video when after building and running the examples in VS2017. Only two applications work: "Hello-world" and "save-to-disk".

Are there any additional requirements that could prevent my computer from starting these applications?
The device manager shows the RGB and the Depth camera (as expected since two applications work fine). 

Do you have any suggestions?

RealSense-Customer-Engineering, 2018-06-12:
[Realsense Customer Engineering Team Comment]
hi @Julz-s,

would like to confirm if pre-built realsense viewer can work on your PC? or only from your build?

RealSense-Customer-Engineering, 2018-06-12:
[Realsense Customer Engineering Team Comment]
Hi @Julz-s,
if the pre-build viewer executable download from GitHub also have the same issue,
please see if your graphic card driver already update to the latest and support openGL.
We use some opengl api for image rendering.
best

RealSense-Customer-Engineering, 2018-06-12:
[Realsense Customer Engineering Team Comment]
Hi @Julz-s,
please discard previous comment about openGL support,

if the pre-build viewer executable download from GitHub also have the same issue,
please see if upgrading your graphic card's driver can help or not
We have customer has similar issue and issue fixed after upgrading the graphic card driver.
best

Julz-s, 2018-06-12:
Hi,
thank you for your replies. 
The .exe has the same issue. It does not load the GUI.

I've just checked the Intel driver update tool telling me that all drivers (except one Bluetooth driver) are up to date.

Best

dorodnic, 2018-06-12:
I don't believe this can be related to drivers. 
Do you have additional USB3 ports on your platforms (perhaps back / front ports)?
Can you run `rs-enumerate` and post the output?

RealSense-Customer-Engineering, 2018-06-25:
[Realsense Customer Engineering Team Comment]
Hi @Julz-s,
are you able to run the viewer now on either on same PC or another PC?
or provide the output of rs-enumerate?

Julz-s, 2018-06-25:
Hi, 
I use another PC now (and a Jetson Board) and it works fine now.
There is/was no issue with connecting the camera (since I was able to use the "save-to-disk" and the "Hello world" examples). However, all examples and tools using video streams don't work at all.
I have no idea why, but I'm with with my current setup.

RealSense-Customer-Engineering, 2018-06-25:
[Realsense Customer Engineering Team Comment]
Hi @Julz-s
(1)are those two png files created by save-to disk looks valid? (like what you could see on viewer)
rs-save-to-disk-output-Depth.png
rs-save-to-disk-output-Color.png

(2) please see if the setting for permission of app to access camera is on or off
setting >privacy->Camera ->let apps to use my camera

closed 2018-08-10 by RealSense-Customer-Engineering