realsenseai/librealsense #1868: cutting a .bag file

Yonatan-Sade, 2018-06-11:
| Required Info                            |                                                                  |
|---------------------------------|------------------------------------------- |
| Camera Model                          | D400 | 
| Operating System & Version     | Win (8.1/10) | 
| Platform                                    | PC |
| Language                            |  C                          | 

### Issue Description
Sorry for the shortness - Is there any option to cut a .bag file? 
I have an enormous .bag file from my camera (25 Giga) 
And I only to save only 10 second from it to another .bag file.
How can I do that?
I looked over the rs-record-playback example, but it seems that it will work only for streaming from device/sensor but NOT from file.
Any idea?

Thank you very much
Yonatan Sade

GuidoBartoli, 2018-06-11:
Are you talking about **ROS bag files**? In that case, you can use `rosbag play` with `--start` and `--duration` options to replay only a part of the recorded messages.

ev-mp, 2018-06-11:
Hello @Yonatan-Sade , in the latest release 2.12.0 we [added a tool](https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#new-features--improvements) that splits rosbag into separate (single) frames : [rs-convert](https://github.com/IntelRealSense/librealsense/tree/development/tools/convert).
Though this is not precisely the functionality that you're asking for, you still may find it useful.

Yonatan-Sade, 2018-06-14:
Let's say I convert this .bag file into million separated frames.
Is there any way to collapse some specific frames into another (small) .bag file?

jbvo, 2018-06-15:
Not sure if it can be used on Windows easily, but I would try doing this with `rosbag filter`, see more info at http://wiki.ros.org/rosbag/Commandline#filter.

RealSense-Customer-Engineering, 2018-06-25:
[Realsense Customer Engineering Team Comment]
Hi @Yonatan-Sade,

Did you try the suggestiosn from @GuidoBartoli and @jbvo to see if it's helpful?

Yonatan-Sade, 2018-06-25:
1. The @GuidoBartoli  approach is not addressing my need since I don't want to play the file. I want to process a small one.
2. The @jbvo approach should be checked.. (do I need some environment to run rosbag filter?)

jbvo, 2018-06-26:
`rosbag filter` doesn't require an environment, on the wiki that I linked to there's a description on how it's supposed to be used, e.g. which variables are available in the python statement that you provide to it.

Examples:

```
rosbag filter my.bag only-tf.bag "topic == '/tf'"
```

```
rosbag filter my.bag sliced.bag "t.secs > 100 and t.secs < 110"
```

Yonatan-Sade, 2018-06-28:
Does it work in windows?

jbvo, 2018-06-28:
I don't know, I'm only working on Ubuntu. I think theoretically is should be possible because rosbag filter is implemented in python completely, but in not sure if someone is actually distributing it.

RealSense-Customer-Engineering, 2018-07-09:
[Realsense Customer Engineering Team Comment]
Ｈi @Yonatan-Sade,

Does it work for you per @jbvo's feedback?

Yonatan-Sade, 2018-07-09:
Did not try (I have only windows)
Meanwhile, we moved to ZED..

closed 2018-07-09 by Yonatan-Sade

jfcisneros, 2018-12-03:
Hi I tried to trim the rosbag using:
https://github.com/IntelRealSense/librealsense/issues/1868#issuecomment-400223270
But when I try to replay it using the realsense-viewer I get the following error:
"Failed to load file /media/.../1_trim.bag. Reason: Failed to create ros reader: Invalid format, file does not contain topic "/file_version" nor "FILE_VERSION"

DSDev-NickHogle, 2020-01-09:
@jfcisneros: I got this working as described in [#1868 (comment)](https://github.com/IntelRealSense/librealsense/issues/1868#issuecomment-400223270) by digging through the `.bag` files that are generated by RealSense Viewer.  It looks like RealSense stores important metadata in some of the messages. These messages all have a timestamp with `secs` = 0 and `nsecs` = 1.  

For example, I was able to generate a trimmed `.bag` file that is usable by RealSense Viewer using this command:
```
rosbag filter my.bag sliced.bag "(t.secs==0 and t.nsecs==1) or (t.secs>=10 and t.secs<=20)"
```

You can see all these topics by using the `-p ` option with rosbag.  i.e.:
```
rosbag filter -p '"%4d.%09d: %s" % (t.secs, t.nsecs, topic)'  my.bag test.bag "(t.secs==0 and t.nsecs == 1)"
```
... resulting in something like the following output.  Note at the top, we see that all-important `/file_version` topic:
```
MATCH    0.000000001: /file_version
MATCH    0.000000001: /device_0/sensor_0/Depth_0/info
MATCH    0.000000001: /device_0/sensor_0/Depth_0/info/camera_info
MATCH    0.000000001: /device_0/sensor_0/Infrared_1/info
MATCH    0.000000001: /device_0/sensor_0/Infrared_1/info/camera_info
MATCH    0.000000001: /device_0/sensor_0/Infrared_2/info
MATCH    0.000000001: /device_0/sensor_0/Infrared_2/info/camera_info
MATCH    0.000000001: /device_0/info
MATCH    0.000000001: /device_0/info
MATCH    0.000000001: /device_0/info
MATCH    0.000000001: /device_0/info
MATCH    0.000000001: /device_0/info
MATCH    0.000000001: /device_0/info
MATCH    0.000000001: /device_0/info
MATCH    0.000000001: /device_0/info
MATCH    0.000000001: /device_0/info
MATCH    0.000000001: /device_0/sensor_0/info
MATCH    0.000000001: /device_0/sensor_0/option/Exposure/value
MATCH    0.000000001: /device_0/sensor_0/option/Exposure/description
MATCH    0.000000001: /device_0/sensor_0/option/Gain/value
MATCH    0.000000001: /device_0/sensor_0/option/Gain/description
MATCH    0.000000001: /device_0/sensor_0/option/Enable Auto Exposure/value
MATCH    0.000000001: /device_0/sensor_0/option/Enable Auto Exposure/description
MATCH    0.000000001: /device_0/sensor_0/option/Visual Preset/value
MATCH    0.000000001: /device_0/sensor_0/option/Visual Preset/description
MATCH    0.000000001: /device_0/sensor_0/option/Laser Power/value
MATCH    0.000000001: /device_0/sensor_0/option/Laser Power/description
MATCH    0.000000001: /device_0/sensor_0/option/Emitter Enabled/value
MATCH    0.000000001: /device_0/sensor_0/option/Emitter Enabled/description
MATCH    0.000000001: /device_0/sensor_0/option/Frames Queue Size/value
MATCH    0.000000001: /device_0/sensor_0/option/Frames Queue Size/description
MATCH    0.000000001: /device_0/sensor_0/option/Asic Temperature/value
MATCH    0.000000001: /device_0/sensor_0/option/Asic Temperature/description
MATCH    0.000000001: /device_0/sensor_0/option/Error Polling Enabled/value
MATCH    0.000000001: /device_0/sensor_0/option/Error Polling Enabled/description
MATCH    0.000000001: /device_0/sensor_0/option/Projector Temperature/value
MATCH    0.000000001: /device_0/sensor_0/option/Projector Temperature/description
MATCH    0.000000001: /device_0/sensor_0/option/Output Trigger Enabled/value
MATCH    0.000000001: /device_0/sensor_0/option/Output Trigger Enabled/description
MATCH    0.000000001: /device_0/sensor_0/option/Depth Units/value
MATCH    0.000000001: /device_0/sensor_0/option/Depth Units/description
MATCH    0.000000001: /device_0/sensor_0/option/Stereo Baseline/value
MATCH    0.000000001: /device_0/sensor_0/option/Stereo Baseline/description
MATCH    0.000000001: /device_0/sensor_0/option/Inter Cam Sync Mode/value
MATCH    0.000000001: /device_0/sensor_0/option/Inter Cam Sync Mode/description
MATCH    0.000000001: /device_0/sensor_0/option/Emitter On Off/value
MATCH    0.000000001: /device_0/sensor_0/option/Emitter On Off/description
MATCH    0.000000001: /device_0/sensor_0/option/Global Time Enabled/value
MATCH    0.000000001: /device_0/sensor_0/option/Global Time Enabled/description
MATCH    0.000000001: /device_0/sensor_0/post_processing
MATCH    0.000000001: /device_0/sensor_0/post_processing
MATCH    0.000000001: /device_0/sensor_0/post_processing
MATCH    0.000000001: /device_0/sensor_0/post_processing
MATCH    0.000000001: /device_0/sensor_0/post_processing
MATCH    0.000000001: /device_0/sensor_0/post_processing
MATCH    0.000000001: /device_0/sensor_0/post_processing
MATCH    0.000000001: /device_0/sensor_1/info
[... 27 lines cut ...]
MATCH    0.000000001: /device_0/sensor_1/option/Power Line Frequency/description
MATCH    0.000000001: /device_0/sensor_1/option/Auto Exposure Priority/value
MATCH    0.000000001: /device_0/sensor_1/option/Auto Exposure Priority/description
MATCH    0.000000001: /device_0/sensor_1/option/Global Time Enabled/value
MATCH    0.000000001: /device_0/sensor_1/option/Global Time Enabled/description
MATCH    0.000000001: /device_0/sensor_1/post_processing
MATCH    0.000000001: /device_0/sensor_2/info
MATCH    0.000000001: /device_0/sensor_2/option/Frames Queue Size/value
MATCH    0.000000001: /device_0/sensor_2/option/Frames Queue Size/description
MATCH    0.000000001: /device_0/sensor_2/option/Global Time Enabled/value
MATCH    0.000000001: /device_0/sensor_2/option/Global Time Enabled/description
MATCH    0.000000001: /device_0/sensor_0/Infrared_1/tf/0
MATCH    0.000000001: /device_0/sensor_0/Infrared_2/tf/0
MATCH    0.000000001: /device_0/sensor_0/Depth_0/tf/0
[... cut ...]
NO MATCH    0.000000600: /device_0/sensor_0/Infrared_1/image/metadata
NO MATCH    0.000000600: /device_0/sensor_0/Infrared_1/image/metadata
NO MATCH    0.000000600: /device_0/sensor_0/Infrared_1/image/metadata
NO MATCH    0.000000600: /device_0/sensor_0/Infrared_1/image/metadata
NO MATCH    0.000000600: /device_0/sensor_0/Infrared_1/image/metadata
NO MATCH    0.000000600: /device_0/sensor_0/Infrared_1/image/metadata
NO MATCH    0.000000600: /device_0/sensor_0/Infrared_1/image/metadata
[etc...]
```

julienguegan, 2022-05-04:
@DSDev-NickHogle I did exactly as you said but I still can't read it using the intelrealsense SDK (cf images below). But note that I can open it using rqt_bag

[image]

[image]

DSDev-NickHogle, 2022-05-13:
> @DSDev-NickHogle I did exactly as you said but I still can't read it using the intelrealsense SDK (cf images below). But note that I can open it using rqt_bag
@julienguegan : Hmm.. Do you think the `HDR Merge filter` error message has something to do with it?  Looking at [the code](https://github.com/IntelRealSense/librealsense/blob/c94410a420b74e5fb6a414bd12215c05ddd82b69/src/proc/hdr-merge.cpp#L35), that warning is generated if there is metadata missing... Perhaps you'd need to also include those `.../metadata` frames too?

Maybe something like this would work? (Note, I have not tested this):

```
rosbag filter my.bag sliced.bag "(t.secs==0 and t.nsecs==1) or (t.secs>=10 and t.secs<=20) or topic.endswith('/metadata')"
```

temav, 2022-10-18:
Hi all!
Sorry to bother, are there any updates? Can anyone explain how to cut bag file with ability to open it in the IntelRealSense Viewer?
I've tried the command below but I got the same error "Failed to read next frame ..." as [comment](https://github.com/IntelRealSense/librealsense/issues/1868#issuecomment-1117436211)
`rosbag filter my.bag sliced.bag "(t.secs==0 and t.nsecs==1) or (t.secs>=10 and t.secs<=20)"`

Adding `or topic.endswith('/metadata')` just copies the initial file

antoniogrimalt, 2024-06-05:
> @DSDev-NickHogle I did exactly as you said but I still can't read it using the intelrealsense SDK (cf images below). But note that I can open it using rqt_bag
> 
> [image]
> 
> [image]

I have the same problem.

Did you find any solution?

MartyG-RealSense, 2024-06-06:
Hi @antoniogrimalt  The RealSense SDK uses a bag format that is based on the ROS rosbag standard but has some differences - as described at https://github.com/IntelRealSense/librealsense/issues/3020#issuecomment-452321766 - and so a rosbag recorded in ROS with rosbag record may not be able to be played back in the RealSense Viewer.

RhinoBlindado, 2024-07-30:
Hello, I also needed this.

I've found this thread and the findings here helped me come up with a script that actually works on the Realsense Viewer v2.55.1, it works with recordings made by the RealSense D455 with firmware version 5.16.0.1 (the only camera I have) either by the Python API or the RealSense app.

[image]

The way I made this work is by using the Python ROS modules ( `rospag`. `rospy`), I figured out that plainly reading the bag and saving it ( `read_messages` and `write` )  even including the header topics would result in an incompatible RS bag probably because some encoding error I figured. Thankfully there exists a `raw` flag, setting it seems to do the trick.

I've written a small working Python script [here](https://gist.github.com/RhinoBlindado/4550691ded75570d6d0ee82d8f90229d) that trims the RealSense bag from an RGB frame number to another frame, or either the start of the recording to a given frame or a given frame to the end of the recording.

It plays on the RS app although the controls bug out a bit: The timeline doesn't go all the way to the real start or end of the recording, but if you let it play it, it plays from the exact frames. Maybe there's another ROS topic that saves the lenght of the original file and it bugs out, I haven't found it. In the Python API it seems to play perfectly.

Hope this is useful to someone in the future.

Also, It'd be great if the RealSense team would consider adding this functionality (trimming a RS bag) to the API, since it's quite convenient in many scenarios.

MartyG-RealSense, 2024-07-30:
Thanks so much @RhinoBlindado for sharing your experience and your Python script at https://gist.github.com/RhinoBlindado/4550691ded75570d6d0ee82d8f90229d

antoniogrimalt, 2024-08-27:
> The way I made this work is by using the Python ROS modules ( `rospag`, `rospy`), I figured out that plainly reading the bag and saving it ( `read_messages` and `write` ) even including the header topics would result in an incompatible RS bag probably because some encoding error. Thankfully there exists a `raw` flag, setting it seems to do the trick.
> 
> I've written a small working Python script [here](https://gist.github.com/RhinoBlindado/4550691ded75570d6d0ee82d8f90229d) that trims the RealSense bag from an RGB frame number to another frame, or either the start of the recording to a given frame or a given frame to the end of the recording.
> 
> It plays on the RS app although the controls bug out a bit: The timeline doesn't go all the way to the real start or end of the recording, but if you let it play it, it plays from the exact frames. Maybe there's another ROS topic that saves the length of the original file and it bugs out, I haven't found it. In the Python API it seems to play perfectly.

Thank you very much @RhinoBlindado for suggesting the use of the raw parameter, with it I was able to solve the problem I had with newer SDK versions.

The problem you have with the timeline and its controls in the RealSense Viewer application is due to the fact that you leave a message gap between the configuration messages published in the first nanosecond (`header_time` in your script) and the first sensor message you decide to save in the trimming. You can see this gap if you open the trimmed .bag in the [rqt_bag](https://wiki.ros.org/rqt_bag) application, which you can run in your ROS noetic environment.

To correct this gap, what I've done is replace the publication time `t` of the first message I want to keep with the value 20000 ns, which is an average publication time that I've been able to observe in other .bags recorded with a D435i. The rest of the messages after the first one have this offset applied so that they keep the original separation. Here is a snippet from my code:

```
BAG_SENSORS_START_TIME = Time(nsecs=20000)  # Publish start time of sensors
...
    input_bag = rosbag.Bag(input_bag_path, "r")
...
        clip_start_time = None

        for topic, msg_raw, t in input_bag.read_messages(raw=True):
...
                if clip_start_time is None:
                    clip_start_time = t

                output_bag.write(
                    topic=topic,
                    msg=msg_raw,
                    t=(t - clip_start_time + BAG_SENSORS_START_TIME),
                    raw=True,
                )
```

In a few weeks I will publish my thesis and in it the rest of the code that I have used to solve what I mentioned.

RhinoBlindado, 2024-09-04:
You're welcome @antoniogrimalt, I'm glad I was able to help. Good luck with the publication of your thesis.

And thank you for pointing out the gap in the bag, I'm still fairly new to the ROS ecosystem. I'm adding your change to my code too.