realsenseai/librealsense #1869: List with all recognized JSON parameters

GuidoBartoli, 2018-06-11:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.9.11.0 | 
| Operating System & Version |  Linux Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.13.0-43)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.12                         | 
| Language                            |  C++                         | 

In [this page](https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets) I read about the JSON visual presets that can be loaded by the SDK.

I would like to create a custom JSON file, but I cannot find a reference for _all_ the parameters recognized by the C++ API, since they are only partially set in every preset examples.
I tried to dig into the SDK code and I found `load_json()` function, but I could not find a comprehensive list.

Thanks

RealSense-Customer-Engineering, 2018-06-12:
[Realsense Customer Engineering Team Comment]
hi @GuidoBartoli,

please refer to #1723, and I think you can find the list in
https://github.com/IntelRealSense/librealsense/blob/master/src/ds5/advanced_mode/json_loader.hpp#L291
please let me know if this is what you are looking for.

GuidoBartoli, 2018-06-12:
Yes, that's it, thank you very much!

closed 2018-06-12 by GuidoBartoli