realsenseai/librealsense #1870: How to extract RGB from point?

Chimdi511, 2018-06-11:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.09.11.00  | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC |
| SDK Version                           |   2.0                         | 
| Language                               |  C/pcl                         | 

I know how to project a point to a pixel. But how do I extract the rgb values from this pixel? I would like to use the Cloud XYZRGB type in PCL, but I need rgb values for the points. 

Thank you!

dorodnic, 2018-06-12:
Hi @Chimdi511 
Please take a look at #1601

Chimdi511, 2018-06-12:
Thank you very much! Just wanted to note that the values returned are ints in the range of 0..255, not floats from 0..1.

closed 2018-06-12 by Chimdi511

Resays, 2019-06-10:
Here is an example to get RGB data from point [how to get XYZ RGB data from RealSense](https://github.com/Resays/xyz_rgb_realsense/tree/master)