realsenseai/librealsense #1873: Frame did not arrive in time with Multiple cameras

manyaafonso, 2018-06-12:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Operating System & Version |  Win 10 | 
| Platform                                 | PC |
| SDK Version                            | 2.10.4    | 
| Language                            |  C# | 

### Issue Description
We are trying to acquire single shot images for 4 connected realsense D435 cameras. We found that streaming all together with high res is hardly possible, even at minimum frame rate. The approach we use is to start a cam, wait for the next frame (color and depth) and stop the cam and this for each cam one after another. With 3 cameras it is working fine but with 4 cams we get the error “Frame did not arrive in time”.
Is this the right approach to take single shots and how can we solve this problem ?
 
Below is our code in C#:
 
for (int i=0; i<4; i++)
{
colorSensor[i].Start(colorQue[i]);
depthSensor[i].Start(depthQue[i]);
  
colorFrame[i] = colorQue[i].WaitForFrame() as VideoFrame;
depthFrame[i] = depthQue[i].WaitForFrame() as VideoFrame;
   
// here we get the error “Frame did not arrive in time!\r\nrs2_wait_for_frame(queue:0000014370F7BAF0)”   
 
colorSensor[i].Stop();
depthSensor[i].Stop();
}

RealSense-Customer-Engineering, 2018-06-26:
[Realsense Customer Engineering Team Comment]
@manyaafonso
I've reduplicated your issue, I encounter the same behavior as yours. But I found out that may come from the powered USB 3.0 hub can't deliver enough power for each device, you can try to connect three camera through the powered USB 3.0 hub then connect the other one through another USB 3.0 port. It works from my end.

Also you can take another test can prove my statement, you can put the open/close sensor inside the for loop, which tune on/off only one camera in each loop. In this case, you can have four cameras connect on the powered USB 3.0 hub without getting any exception, since every only one camera consume the power. Attached the code in below comment.

carypai, 2018-06-26:
[Program.zip](https://github.com/IntelRealSense/librealsense/files/2135912/Program.zip)

manyaafonso, 2018-06-26:
Hi, thanks for your reply and for the code. We will try it out and inform you.

manyaafonso, 2018-07-05:
@RealSense-Customer-Engineering @carypai 
We have tried your suggestion and included code but it is not practical for us. Opening and closing each camera every time is taking way too much time. For 3 cameras this is over 13 seconds.

RealSense-Customer-Engineering, 2018-07-06:
[Realsense Customer Engineering Team Comment]
@manyaafonso
As I mentioned, this is power not enough issue if you want to use existing environment to have four cameras working, you need to close and open device. If you don't want to turn off the camera, you can consider to plug in the forth camera to another USB controller OR increase the adapter output of USB 3.0 powered hub with output 2A(suggestion).

Below is the rough calculation :
When device is not streaming but sensor is opened, roughly consuming : 0.2W
Maximum Streaming (with highest resolution) : 2W
ASIC board static consuming : 1.306

with four camera connected to hub (1 streaming the other 3 with open state) consuming :
1.5*3 + 3.3 = 7.8W / 5V = 1.56A

RealSense-Customer-Engineering, 2018-07-12:
[Realsense Customer Engineering Team Comment]
@manyaafonso
Any update?

closed 2018-07-23 by RealSense-Customer-Engineering