realsenseai/librealsense #1875: Mac: Duplicated serials when putting two D435 on one USB hub

emilk, 2018-06-12:
When I plug in two D435 cameras into two separate USB holes on my MacBook:

```
> ./rs-enumerate-devices | grep "Device info" -A 9
 Device info: 
    Name                          : 	Intel RealSense USB2
    Serial Number                 : 	733512070092
    Firmware Version              : 	05.08.15.00
    Recommended Firmware Version  : 	05.09.09.02
    Physical Port                 : 	
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0AD6

--
 Device info: 
    Name                          : 	Intel RealSense USB2
    Serial Number                 : 	750612070991
    Firmware Version              : 	05.08.15.00
    Recommended Firmware Version  : 	05.09.09.02
    Physical Port                 : 	
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0AD6

``` 

When I plug in the same cameras in one USB hub, and plug that USB hub into my MacBook:

```
> ./rs-enumerate-devices | grep "Device info" -A 9
 Device info: 
    Name                          : 	Intel RealSense D435
    Serial Number                 : 	733512070092
    Firmware Version              : 	05.08.15.00
    Recommended Firmware Version  : 	05.09.09.02
    Physical Port                 : 	
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B07

--
 Device info: 
    Name                          : 	Intel RealSense D435
    Serial Number                 : 	733512070092
    Firmware Version              : 	05.08.15.00
    Recommended Firmware Version  : 	05.09.09.02
    Physical Port                 : 	
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B07
```

Note the duplicated serial numbers.

The USB hub in question is a SIIG Industrial USB 3.0: http://www.siig.com/it-products/usb/converters/4-port-industrial-usb-to-rs-232-serial-with-15kv-esd.html

The behavior is not perfectly predictable – sometimes I get two separate serials via the hub, sometimes not.

dorodnic, 2018-06-12:
We will keep investigating.

RealSense-Customer-Engineering, 2018-06-13:
[Realsense Customer Engineering Team Comment]
hi @emilk,

I don't have same hub with you, but let me try to reproduce first, and you say sometimes, how frequency it is? like 10% or 30%?
thanks.

mattfelsen, 2018-06-13:
I'm seeing the same behavior, except it's not limited to having the cameras on the same hub. I see the same behavior with different configurations, such as:
- one camera directly connected, one on a hub
- both cameras directly connected but on separate busses

On the MacBook Pros with all USB-C ports, it seems like each side of the laptop is a separate bus

Here are some screenshots showing what `rs-enumerate-devices` outputs along with what I'm seeing in Sys Info. Also of note is that the serial in Sys Info is different from what `rs-enumerate-devices` shows, but I'm not sure if that is a problem or not.

One directly connected, one on a USB hub:
[image]

Both cams directly connected on opposite sides of the laptop:
[image]

RealSense-Customer-Engineering, 2018-07-12:
[Realsense Customer Engineering Team Comment]
hi @mattfelsen, @emilk,

Pull Request #2039 created for this issue, please let me know how it works at your site.
thanks.

closed 2018-07-25 by RealSense-Customer-Engineering