realsenseai/librealsense #1877: D435 with Python 3.6.4 & Windows 10 64Bit > DLL load failed

LeMajors, 2018-06-12:
Camera Model: RealSense D435
Firmware Version: 05.09.11.00
PC, Windows 10 64Bit
SDK 2.0
Python 3.6.5

Since it was mentioned in previous posts that pyrealsense2 will also be made available in the future with pip install ... for Python 3.x, I have tried this.

The installation ended successfully:
Successfully installed pyrealsense2-2.9.1

However, example scripts stop with an import error:
```Traceback (most recent call last):
  File "C:\Arbeit\Python Tests\realsense\run.py", line 12, in <module>
    import pyrealsense2 as rs
  File "C:\Users\xxxxx\AppData\Local\Programs\Python\Python36\lib\site-packages\pyrealsense2\__init__.py", line 2, in <module>
    from .pyrealsense2 import *
ImportError: DLL load failed: Das angegebene Modul wurde nicht gefunden.
```
... although two new folders have been created in "../Programs/Python/Python36/Lib/site-packages/":
- pyrealsense2 (containing "realsense2.dll")
- pyrealsense2-2.9.1.dist-info

My Questions:
- Is this workflow not yet implemented for Python 3.x?
- If not - is it possible to get the files "realsense2.dll" and "pyrealsense2.pyd" für Python 3.6.5 & Win10 64Bit as download. (I have not been able to compile the source code myself, unfortunately I'm very inexperienced in this area and I would need a much more detailed tutorial than what have posted so far.)

dorodnic, 2018-06-13:
Hi @LeMajors 
Can you try the files installed with [Intel.RealSense.SDK.exe](https://github.com/IntelRealSense/librealsense/releases/download/v2.12.0/Intel.RealSense.SDK.exe)? Under `C:\Program Files\Intel RealSense SDK 2.0\bin` we provide x86 and x64 versions for both python 2.7 and 3.6.
We are also planning to expand the list of versions available via pip, but it is still in early stages right now.

kargarisaac, 2018-06-13:
Is there any tutorial or pipeline to install all requirement packages for working with realsense d400 data in windows10 and python?
I completely confused about what to do. 
how can I get video frames online?
how can I read saved .bag files in windows 10?

Thanks

dorodnic, 2018-06-13:
Hi @kargarisaac 

* We have some documentation here: [wrappers/python](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python)
* We are trying to set-up some web-based demos, but they are not ready yet.
* Working from bag file should be relatively easy:
```python
import pyrealsense2 as rs
pipeline = rs.pipeline()
cfg = rs.config()
cfg.enable_device_from_file("recording.bag")
pipeline.start(cfg)
# work with the pipeline as if it was a real device
```

kargarisaac, 2018-06-13:
you mentioned in the python wrapper directory that:
Note:

pyrealsense AKA pyrealsense/2.0 is a community supported Python wrapper for librealsense v1.12.1, This wrapper does not support newer versions and does not work with the RealSense SDK 2.0

I have a d415 camera that work with sdk 2.0. Can I use python 3?

kargarisaac, 2018-06-13:
Thanks. I installed and got the images.

Another question:
why there is a shadow on the left side of depth images in realsense viewer? 
Is there any forum I can ask my question?

Thanks

LeMajors, 2018-06-13:
Hi @dorodnic 
thanks for you answer.  In "C:\Program Files (x86)\Intel RealSense SDK 2.0\bin\x64" I find
- Intel.Realsense.dll (only 56KB)
- pyrealsense2.cp36-win_amd64.pyd (3.298 KB)
- pyrealsense2.pyd

Where do I put these Files? At "C:\Users\xxx\AppData\Local\Programs\Python\Python36\Lib\site-packages\pyrealsense2" I find on similar name and some different files:
- pybackend2.cp36-win_amd64.pyd
- pyrealsense2.cp36-win_amd64.pyd (8.046 KB)
- realsense2.dll (19.589 KB > much bigger than the dll above)

0xLeon, 2018-06-18:
See #1907 for a solution on the original problem.

Regarding your question placement of the libraries: I wouldn't mess with the Python directory itself. The better option would be to place the binaries somewhere on your disk and then point to this directory via the PYTHONPATH environment variable. This way, yo can keep your Python installation directory clean. For Python 3, you would need to place the two dll files as well as the cp36 pyd file in this directory.

RealSense-Customer-Engineering, 2018-06-22:
[Realsense Customer Engineering Team Comment]
Hi @LeMajors,

Still need any further support required for this issue?

LeMajors, 2018-06-22:
At the moment I'm abroad and can't test 0xLeons hint.
So first of all: thanks @0xLeon for your help. I will test this in about 2 weeks.
@RealSense-Customer-Engineering: You can close this issue. If it's not working, I will open a new one.

Regards, Marcus

closed 2018-06-25 by RealSense-Customer-Engineering