realsenseai/librealsense #1882: Unique ID of stream profile is not unique

sogartar, 2018-06-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.9.11.0 | 
| Operating System & Version |   Linux (Ubuntu 16.04)  | 
| Kernel Version (Linux Only)    |  4.10.0                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.12.0                          | 
| Language                            |  C++                         | 

### Issue Description
I have something like:
```
rs::sensor sensor = ...
auto stream_profiles = sensor.get_stream_profiles();
```
For example these two profiles do not have a unique ID.

| | | |
|-|-|-|
| index | 1 | 1 |
| uid | 5 | 5 |
| framerate | 30 | 25 |
| format | RS2_FORMAT_Y8 | RS2_FORMAT_Y16 |
| type | RS2_STREAM_INFRARED | RS2_STREAM_INFRARED |

I expect them to be unique. What is the scope of uniqueness? It should be at least the device. Better yet across all connected devices.

dorodnic, 2018-06-13:
Scope of uniqueness is within streaming session (across all devices). 
Streams with the same unique ID cannot co-exists (due to hardware limitations). 
This is how the Viewer knows to aggregate stream profiles into groups that can be turned-on together. 
While streaming, the unique ID is useful to address `frame_queues` or rendering buffers, or any other resource you have once-per-stream.

sogartar, 2018-06-14:
This is rather misleading. So it is some sort of ID of a stream profile group, where only one stream can be active from the group.

closed 2018-06-14 by sogartar

sogartar, 2018-06-14:
If you create a new rs2::context, each time the streams would have different IDs.