realsenseai/librealsense #1884: Multiple camera hardware sync: no master signal

nielsgeens, 2018-06-13:
| Required Info                         |                                                        |
|---------------------------------|------------------------------------- |
| Camera Model                       | D435                                              | 
| Firmware Version                   | D400 5.9.11                                    | 
| Operating System & Version |   Win 10                                         | 
| Platform                                 | PC                                                  |
| SDK Version                            | RealSense SDK 2.0 build 2.12.0     | 
| Language                               |  C++                                              | 

### Issue Description
I'm trying to hardware sync two D435 cameras as is explained in the [Multiple Camera WhitePaper](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense_Multiple_Camera_WhitePaper.pdf). The trigger pin of the camera configured as master stays floating and does not generate the trigger signal I would expect. I'll paste the relevant code below and attach the full code as zip. I'm pretty sure I'm overlooking something but I can't seem to find more information online.

Any help would be appreciated.

Niels

```

const size_t syncLength = 24;
const uint8_t syncDefault[syncLength]  = {0X14, 0X00, 0XAB, 0XCD, 0X64, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0};
const uint8_t syncMaster[syncLength]  = {0X14, 0X00, 0XAB, 0XCD, 0X64, 0, 0, 0, 0X01, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0};
const uint8_t syncSlave[syncLength]  = {0X14, 0X00, 0XAB, 0XCD, 0X64, 0, 0, 0, 0X02, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0};
```

```
    cout << "Setting up hardware trigger: ";
    bool master = false;
    bool errors = false;
    for (auto device: devices) {
        const shared_ptr<rs2_device> d = device.get();
        if (!d) {
            errors = true;
            cout << "No rs2_device. " << endl;
            continue;
        }
        rs2_error* e = nullptr;
        const rs2_raw_data_buffer* receiveBuffer;
        if (!master) {
            receiveBuffer = rs2_send_and_receive_raw_data(d.get(), (void*)syncMaster, syncLength, &e);
            master = true;
        } else {
            receiveBuffer = rs2_send_and_receive_raw_data(d.get(), (void*)syncSlave, syncLength, &e);
        }
        if (e) {
            errors = true;
            cout << rs2_get_error_message(e) << " ";
            rs2_free_error(e);
            e = nullptr;
        }
        rs2_delete_raw_data(receiveBuffer);
    }
    if (errors) {
        cout << endl;
    } else {
        cout << "Ok" << endl;
    }
```

[main.zip](https://github.com/IntelRealSense/librealsense/files/2098864/main.zip)

ankarako, 2018-06-14:
@nielsgeens  I think that this is a FW issue... Multi-Camera HW sync is not really yet available by the current FW

nielsgeens, 2018-06-14:
Oh, thanks for letting me know. Do you know if there is a timeline somewhere for the firmware development? Or a list of implemented and in progress features?

ffrog12, 2018-06-14:
@nielsgeens It is for Q2 18 apparently but I haven't seen any update on that. 

https://realsense.intel.com/wp-content/uploads/sites/63/BKMs-For-Tuning-RealSense_D4xx_Cameras_WP_1.7.pdf

nielsgeens, 2018-06-14:
Thanks for the info! I've noticed the same question has been asked here before, so I'll close this issue and follow up on one of the others.

closed 2018-06-14 by nielsgeens

ankarako, 2018-06-14:
Last time I asked at Intel RealSense Zendesk (I have asked a lot of times) they said that it is programmed for Q3 2018

ffrog12, 2018-06-14:
@ankarako Very disappointing, if you sign an NDA can you get hold of it early?

ankarako, 2018-06-14:
@ffrog12 I don't think so but I am not really sure either...