realsenseai/librealsense #1885: query_devices() hangs on Mac

mattfelsen, 2018-06-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415, D435 | 
| Firmware Version                   | 05.09.11.00 | 
| Operating System & Version |   MacOS 10.13.4  | 
| Platform                                 | PC |
| SDK Version                            |  2.11.1, 2.12.0                          | 
| Language                            |  C++                          | 

### Issue Description
I'm seeing various types of crashes & hangs on Mac. Occasionally `pipeline.start()` after having previously stopped will crash, but the most reproducible issue is that `context.query_devices()` will hang. This usually happens if it's called after `pipeline.stop()`, which is sometimes solved by sleeping for 1sec first after stopping the pipeline. In addition, plugging in a camera will sometimes cause the Viewer app to freeze. When I look at where it's paused, it's the same place in both my app and the Viewer, which is somewhere in libuvc when trying to access `rs2::device`s returned from `query_devices()`. Note that the call to `query_devices()` returns -- it looks like it's when accessing the items inside the `rs2::device_list` that causes the issue.

Here are some screenshots of the stack trace in both my app and the RS Viewer.

[image]

[image]

RealSense-Customer-Engineering, 2018-06-15:
[Realsense Customer Engineering Team Comment]
hi @mattfelsen,

issue seems same to #1855, thank you for detail,  we are looking into this issue.

RealSense-Customer-Engineering, 2018-07-17:
[Realsense Customer Engineering Team Comment]
hi @mattfelsen,

pull request 2022 should fix this issue, please let me know how it works, thanks.