realsenseai/librealsense #1889: "No Frames Received!" on RealSense Viewer on Ubuntu 16.04

tenmillion, 2018-06-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.09.11.00 | 
| Operating System & Version |   Linux (Ubuntu 16.04)  | 
| Kernel Version (Linux Only)    |  4.13.0-45-generic               | 
| Platform                                 | PC  |
| SDK Version                            |  2.0 (Only using RealSense Viewer v2.12.0 atm)   | 

### Issue Description
I checked some similar issues (https://github.com/IntelRealSense/librealsense/issues/901, https://communities.intel.com/thread/122871, https://github.com/IntelRealSense/librealsense/issues/927) but wasn't able to resolve mine, so I'm posting a new one.

I've just received three D435 cameras and I've been testing them out with RealSense Viewer. I'm using USB 3.0 (might be 3.1, not sure. They show up as USB 3.1 on the Viewer) to connect all three cameras, and I'm using the 1m passive USB 3.0 cables that came with them.

The stereo modules have been pretty unreliable, and I'm wondering how I can make them more reliable.

What's happening is that all the modules were working fine one minute, but when I closed and started up RealSense Viewer again, all the depth streams were blank, and it said "No Frames Received!" The IR and RGB streams were fine.
 
**Fig. 1 Three cameras working fine (but see "incomplete frame detected"):**
[image]

**Fig. 2 Depth stream gone even at lowest resolution & frame rate:**
[image]

After a reboot and another relaunch of RealSense Viewer, one of the depth streams started working again, but the other two were still. (I've also noticed that the framerate is sometimes unstable, but that might not be related to this issue.)

**Fig. 3 One camera came back but two are still gone:** 
[image]

The CPU doesn't seem _that_ busy:
**Fig. 4 Overall CPU usage is 13%:** 
[image]
**Fig. 4 Although one of them is at 100%:** 
[image]

I think just before the depth stream started to act strange, I noticed that the USB connections were showing up as "USB 2.1" (although I didn't change anything about the USB), but I'm not sure and I don't have screenshots of that. 

This is what `udevadm` outputs when I plug in a camera that's working (not sure if this helps, sorry):
> KERNEL[3593.040695] add      /devices/pci0000:00/0000:00:14.0/usb4/4-6 (usb)
> KERNEL[3593.041249] add      /devices/pci0000:00/0000:00:14.0/usb4/4-6/4-6:1.0 (usb)
> KERNEL[3593.042160] add      /devices/pci0000:00/0000:00:14.0/usb4/4-6/4-6:1.0/video4linux/video5 (video4linux)
> KERNEL[3593.042837] add      /devices/pci0000:00/0000:00:14.0/usb4/4-6/4-6:1.0/video4linux/video7 (video4linux)
> KERNEL[3593.042880] add      /devices/pci0000:00/0000:00:14.0/usb4/4-6/4-6:1.0/media3 (media)
> KERNEL[3593.042936] add      /devices/pci0000:00/0000:00:14.0/usb4/4-6/4-6:1.0/input/input20 (input)
> KERNEL[3593.043004] add      /devices/pci0000:00/0000:00:14.0/usb4/4-6/4-6:1.0/input/input20/event18 (input)
> KERNEL[3593.043156] add      /devices/pci0000:00/0000:00:14.0/usb4/4-6/4-6:1.1 (usb)
> KERNEL[3593.043289] add      /devices/pci0000:00/0000:00:14.0/usb4/4-6/4-6:1.2 (usb)
> KERNEL[3593.043419] add      /devices/pci0000:00/0000:00:14.0/usb4/4-6/4-6:1.3 (usb)
> KERNEL[3593.044305] add      /devices/pci0000:00/0000:00:14.0/usb4/4-6/4-6:1.3/video4linux/video8 (video4linux)
> KERNEL[3593.044390] add      /devices/pci0000:00/0000:00:14.0/usb4/4-6/4-6:1.3/media4 (media)
> KERNEL[3593.044528] add      /devices/pci0000:00/0000:00:14.0/usb4/4-6/4-6:1.4 (usb)
> UDEV  [3593.049160] add      /devices/pci0000:00/0000:00:14.0/usb4/4-6 (usb)
> UDEV  [3593.051058] add      /devices/pci0000:00/0000:00:14.0/usb4/4-6/4-6:1.0 (usb)
> UDEV  [3593.052679] add      /devices/pci0000:00/0000:00:14.0/usb4/4-6/4-6:1.2 (usb)
> UDEV  [3593.052873] add      /devices/pci0000:00/0000:00:14.0/usb4/4-6/4-6:1.1 (usb)
> UDEV  [3593.053110] add      /devices/pci0000:00/0000:00:14.0/usb4/4-6/4-6:1.4 (usb)
> UDEV  [3593.053545] add      /devices/pci0000:00/0000:00:14.0/usb4/4-6/4-6:1.3 (usb)
> UDEV  [3593.054877] add      /devices/pci0000:00/0000:00:14.0/usb4/4-6/4-6:1.0/media3 (media)
> UDEV  [3593.055642] add      /devices/pci0000:00/0000:00:14.0/usb4/4-6/4-6:1.3/media4 (media)
> UDEV  [3593.056707] add      /devices/pci0000:00/0000:00:14.0/usb4/4-6/4-6:1.0/video4linux/video5 (video4linux)
> UDEV  [3593.056887] add      /devices/pci0000:00/0000:00:14.0/usb4/4-6/4-6:1.0/video4linux/video7 (video4linux)
> UDEV  [3593.057021] add      /devices/pci0000:00/0000:00:14.0/usb4/4-6/4-6:1.0/input/input20 (input)
> UDEV  [3593.059901] add      /devices/pci0000:00/0000:00:14.0/usb4/4-6/4-6:1.3/video4linux/video8 (video4linux)
> UDEV  [3593.117178] add      /devices/pci0000:00/0000:00:14.0/usb4/4-6/4-6:1.0/input/input20/event18 (input)

It doesn't output anything when I'm just running RealSense Viewer. It also doesn't seem to matter which USB plug a camera is connected to, it's the same ones that don't work no matter where they're plugged in.

I'd appreciate any help with this. Thanks!

abecciu, 2018-06-20:
@RealSense-Customer-Engineering any idea of what could be happening?

It's essentially impossible to use the camera in this state.

Thanks!

RealSense-Customer-Engineering, 2018-06-20:
[Realsense Customer Engineering Team Comment]
Hello and thank you for your inquiry,

"No Frames Received" error is a known issue with RealSense Viewer 2.12.  The next release of RealSense Viewer should address issues seen with 2.12 and multiple cameras.  In the meantime, please use an older version of LRS, such as LRS 2.11.  LRS 2.11 does not have the same issue with multiple cameras that affects 2.12.

LRS 2.11:  https://github.com/IntelRealSense/librealsense/tree/v2.11.0

Thank you.

abecciu, 2018-06-20:
@RealSense-Customer-Engineering thanks for your response. But I have the same issue with all the examples provided, not only with Realsense Viewer. And I have only one camera.

Is the known issue that you mention specific to RealsenseViewer or it's a problem with the drivers?

RealSense-Customer-Engineering, 2018-06-21:
[Realsense Customer Engineering Team Comment]
Hello again abecciu,

We have not been able to replicate this issue on our systems.  Please answer the questions below so we can help narrow down your issue:

-  What kind of system are you using?  Intel UP Board, NUC, Laptop?  Please provide model name and #.

-  Are you plugging in camera to USB 3.0 ports?

-  Have you tried different USB 3.0 type-C cables and plugging those in different ports?

-  Are you using a USB 3.0 powered hub?

I am thinking the issue you are seeing is related to power consumption, could you please:

1. Type `export LRS_LOG_LEVEL=Debug`.
2. Reproduce the issue.
3. To compare camera functionality please also post the output of `rs-enumerate-devices`

Thank you

abecciu, 2018-06-23:
@RealSense-Customer-Engineering  Thanks for the prompt response! Please, find my answers below:

*What kind of system are you using? Intel UP Board, NUC, Laptop? Please provide model name and #.*

**I'm using a PC (amd ryzen threadripper 1950x and x399 motherboard).**

*Are you plugging in camera to USB 3.0 ports?*

**YES**

*Have you tried different USB 3.0 type-C cables and plugging those in different ports?*

**I just tried plugging the camera directly to a usb port on the motherboard and it works correctly now. I don't have this issue anymore.**

*Are you using a USB 3.0 powered hub?*

**Yes, I was using a hub. However, on Windows with the same machine and hub I didn't experience any of these problems. Only on Linux.**

**Would it be possible to make the camera work correctly with a USB hub on Linux? My production application will require that the camera be connected through a hub.**

Thanks so much more your help!

RealSense-Customer-Engineering, 2018-07-16:
[Realsense Customer Engineering Team Comment]
hi @tenmillion,

your case normally due to USB hub power is not sufficient for 3 camera to turn on at the same time, so want to make sure if you are using powered USB hub?

hi @abecciu,
your case maybe an known issue that camera have this kind of issue if after camera improper shutdown, and your workaround also same workaround for this known issue, so glad that you have a workaround so you can continue and we are still fixing this issue.

tenmillion, 2018-07-17:
Hi @RealSense-Customer-Engineering,
Thanks for the reply, and sorry for the delayed reply.
I actually got the cameras to work after changing from the USB ports on the motherboard to an external powered USB hub, so I guess that was it. It still stops working if I raise the resolution above 640 x 480 px at 30 fps, but I presume that's a separate issue, and for now this is good enough for my purposes (although it would be great if higher resolutions are also supported). 

@abecciu FYI my computer is running Linux with an external powered USB hub, but I don't seem to be having the same issue as you, so I'm not sure what was wrong.stillstill occasionally stop getting frames on some of my cameras but this seems to happen after I've changed some settings on realsense viewer (e.g. resolution, haven't quite figure out which ones are buggy yet), and it restarts normally if I just switch off and on the USB hub. As long as I'm just recording frames on my app it keeps going fine.

gongyue666, 2018-09-05:
I face the same problem: first time work, others are not, must reinsert the USB. The USB power is well

RealSense-Customer-Engineering, 2018-09-05:
[Realsense Customer Engineering Team Comment]
Hi @gongyue666,

1. Please use the latest librealsense (ex: 2.16.0) and upgrade your firmware to the latest (ex: 5.10.3.0) and see how it works.
2. Try to have the workaround as hardware reset before stream like below discussion (the code like: dev.hardware_reset() ).
https://communities.intel.com/message/558651

Hi @tenmillion,

It's the bandwidth limitation of one USB hub with only one XHCI controller if a higher-solution streams required. You can refer to the multiple camera white paper to have a better idea.

gongyue666, 2018-09-06:
Thx for your reply, I update fireware and librealsense, the problem still be. And what's worse, when I use ctrl+c to stop the realsense, sometimes the realsense is still running (I see the red light from the sensor)

RealSense-Customer-Engineering, 2018-09-07:
[Realsense Customer Engineering Team Comment]
Hi @gongyue666,

It looks it's a separate issue about this improper shutdown to cause the IR projector still on. Can you file another ticket to track this and describe more detail? Do you use realsense app based on librs 2.16.0 + D435 FW5.10.3? or, which app?

gongyue666, 2018-09-07:
hi, thx for reply,I can realsense app based on librs 2.16.0 + D435 FW5.10.3 as you let me do as last reply. The method that reset the USB port I haven't test till now, I will try today. and what can I give you to the "the IR projector still on" question, any advice can give me or  what logs do you need?

RealSense-Customer-Engineering, 2018-09-07:
[Realsense Customer Engineering Team Comment]
Hi @gongyue666,

For "IR projector still on" issue:
1. You can give us your system configuration (ex: kernel version), which app used and the reproducing steps? How many tries to be able to reproduce this issue?
2. You can also enable the log "export LRS_LOG_LEVEL=DEBUG", the log file will be saved for further debug when you run librealsense apps.
3. Run "usb-devices" to show the usb device connection status

RealSense-Customer-Engineering, 2018-09-26:
[Realsense Customer Engineering Team Comment]
Hi @gongyue666,

Are you still able to reproduce "IR projector still on" issue on the latest librealsense and firmware?

arun-kumark, 2018-10-03:
Hello all,
I am facing the same issue on **realsense-viewer 2.16** on **Ubuntu 18.04 LTS.**

I had tried the patch given at the following link: 
_https://communities.intel.com/thread/125135_

But the camera dies after few minutes yielding the following logs behind:
 ```
03/10 09:48:25,035 ERROR [140425696376576] (error-handling.cpp:66) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Bad file descriptor
 03/10 09:48:26,035 ERROR [140425696376576] (types.h:179) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Bad file descriptor
 03/10 09:48:26,035 ERROR [140425696376576] (error-handling.cpp:66) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Bad file descriptor
 03/10 09:48:27,035 ERROR [140425696376576] (types.h:179) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Bad file descriptor
 03/10 09:48:27,036 ERROR [140425696376576] (error-handling.cpp:66) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Bad file descriptor
 03/10 09:48:28,036 ERROR [140425696376576] (types.h:179) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Bad file descriptor
 03/10 09:48:28,036 ERROR [140425696376576] (error-handling.cpp:66) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Bad file descriptor
 03/10 09:48:29,036 ERROR [140425696376576] (types.h:179) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Bad file descriptor
 03/10 09:48:29,036 ERROR [140425696376576] (error-handling.cpp:66) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Bad file descriptor
 03/10 09:48:30,036 ERROR [140425696376576] (types.h:179) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Bad file descriptor
 03/10 09:48:30,036 ERROR [140425696376576] (error-handling.cpp:66) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Bad file descriptor
 03/10 09:48:31,036 ERROR [140425696376576] (types.h:179) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Bad file descriptor
 03/10 09:48:31,036 ERROR [140425696376576] (error-handling.cpp:66) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Bad file descriptor
 03/10 09:48:32,036 ERROR [140425696376576] (types.h:179) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Bad file descriptor
 03/10 09:48:32,037 ERROR [140425696376576] (error-handling.cpp:66) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Bad file descriptor
 03/10 09:48:33,037 ERROR [140425696376576] (types.h:179) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Bad file descriptor
 03/10 09:48:33,037 ERROR [140425696376576] (error-handling.cpp:66) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Bad file descriptor

```
Please let me know, how to debug this issue.
Kind Regards
arun

RealSense-Customer-Engineering, 2018-10-22:
[Realsense Customer Engineering Team Comment]
Hi @arun-kumark,

After the clarification, do you still have the issue on your system setup?

arun-kumark, 2018-10-22:
yes, the issue was solved by applying the following Driver Patch:

```
"The Linux Firmware Update Tool should work on Ubuntu 18.04 - we have just tested it in the lab. Please follow the steps below from the Linux DFU Guide:

1. echo 'deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial main' | sudo tee /etc/apt/sources.list.d/realsensepublic.list
2. sudo apt-key adv --keyserver keys.gnupg.net --recv-key 6F3EFCDE
3. sudo apt-get update
4. sudo apt-get install intel-realsense-dfu*
5. Download the latest signed fw from: https://downloadcenter.intel.com/download/27522/Latest-Firmwarefor-Intel-RealSense-D400-Product-Family?v=t
6. lsusb

Notice “Intel Corp.” bus and device numbers; DFU tool uses these

values to identify Intel® RealSense™ D400 series camera

7. Type command:

(This command specifies bus #, device #, -f flag to force upgrade, and –i flag for complete system path to downloaded FW.bin file.)

intel-realsense-dfu –b 002 –d 002 –f –i /home/intel/downloads/Signed_Image_UVC_5_10_3_0.bin

8. Check firmware with command:

intel-realsense-dfu –p

intel-realsense-dfu –b 002 –d 002 –f –i /home/arun/Downloads/Signed_Image_UVC_5_10_3_0.bin

arun@arun:~$ intel-realsense-dfu -p
FW version on device = 5.10.3.0
MM FW Version = 255.255.255.255"
```

RealSense-Customer-Engineering, 2018-10-31:
[Realsense Customer Engineering Team Comment]
Hi @tenmillion, @gongyue666,

You can have a try of librealsense 2.16.3 to how it improves for multiple camera streams and check if "no frames recieved" error fixed.

kaixin-bai, 2018-11-08:
wenn the first time i use D435, it works fine, and then, the realsense-reviewer cannot detecte the camera, after i restart the computer, the camera can be detected, but no frames recieved.
08/11 10:11:42,630 WARNING [139648084985600] (backend-v4l2.cpp:1013) Frames didn't arrived within 5 seconds
Ubuntu 16.04
@RealSense-Customer-Engineering 
i was following https://software.intel.com/en-us/realsense/d400/get-started

RealSense-Customer-Engineering, 2018-12-17:
[Realsense Customer Engineering Team Comment]
would you like to have a try of librealsense 2.17.0 + firmware 5.10.13.0 to see how it improves?

RealSenseCustomerSupport, 2019-02-12:
----------------------------------------------

If no further question, will close this.

closed 2019-03-06 by RealSenseCustomerSupport

whatisor, 2019-06-07:
Can we reopen this?
My case is more difficult.
I am using version firmware 5.11.1.100 and SDK 2.17.0 with Ubuntu 16.04.
Scenario:
1. Connect device
2. Start streamer and work well.
3. Disconnect device (Streamer keep working)
4. Reconnect device (Stream try to reconnect successfully(pipeline start without error) but wait_for_frame always get an exception)
And when I check debug log of SDK, I found this "Frames didn't arrived within 5 seconds".
[resultNotsuccess.txt](https://github.com/IntelRealSense/librealsense/files/3264338/resultNotsuccess.txt)

gayanbrahmanage, 2019-06-07:
Update Firmware to [latest](https://downloadcenter.intel.com/download/27514/Windows-Device-Firmware-Update-Tool-for-Intel-RealSense-D400-Product-Family?_ga=2.258438154.1418070014.1559923477-1885791040.1550602723). It is tested and works. In my case, old firmware worked with windows and not with Ubuntu. Updating to the latest solved the problem.

whatisor, 2019-06-09:
@gayanbrahmanage , thank for reply. Have you got the same issue in Ubuntu?
When check Official website, they said Ubuntu is supported long time ago...

gayanbrahmanage, 2019-06-09:
Yes, It supports Ubuntu. I have two of D435s. One of them didn't work with Ubuntu and only worked with  Windows. Once I updated the Firmware, it worked with both OS.

sawyermade, 2019-12-09:
This is still happening. Use it once and it wont work again until you either restart or unplug/replug in the usb cable. WTF is going on?

PAWAN-PRAJAPATI, 2019-12-23:
I also go the same error. One possible solution is to try following command while keeping all realsense connected.
sudo reboot