realsenseai/librealsense #1891: Pixel 2, Android O cannot connect to D415

Performador, 2018-06-14:
Re-openning #1798 --- please see this for the detailed info.

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.09.11.00 | 
| Operating System & Version |   Ubuntu | 
| Kernel Version (Linux Only)    |  4.9.0                                         | 
| Platform                                 | Android  (8.1.0)|
| SDK Version                            |  2.11.1                        | 

Using libUSB, the samples cannot see or connect to the device.
Using libUVC the samples are able to connect and enumerate devices but hang / do not receive data.

RealSense-Customer-Engineering, 2018-07-23:
[Realsense Customer Engineering Team Comment]
@performador Could you please provide the logcat and dmesg for the issue? Thanks!

RealSense-Customer-Engineering, 2018-08-13:
[Realsense Customer Engineering Team Comment]
Have you try other USB3 cable?
BTW, when you using libusb, have you apply the kernel patch into your android kernel? In this way, Android is a special Linux distribution.

RealSense-Customer-Engineering, 2018-08-20:
[Realsense Customer Engineering Team Comment]
@Performador Any update of this issue? Can other USB3 cable work?  Have you applied the kernel patch? Thanks!

RealSense-Customer-Engineering, 2018-08-28:
[Realsense Customer Engineering Team Comment]
@Performador  Any update? Waiting for your reply. Thanks!

RealSense-Customer-Engineering, 2018-09-03:
[Realsense Customer Engineering Team Comment]
Hi Performador,
Thank you for your patch.

I think you have to try other USB OTG adapter, if system said that is USB 2.0 port.

BTW, please make sure the resolution and fps are existed in the output of "rs-enumerate-devices -m"
https://github.com/IntelRealSense/librealsense/blob/master/doc/android/AndroidNativeSamples.md

closed 2018-09-18 by RealSense-Customer-Engineering