realsenseai/librealsense #1892: Build an Android Application for RealSense SDK  question

yingthink, 2018-06-15:
Required   Info |  **``**
-- | --
Camera Model | D415
Firmware Version |  
Operating System & Version |  ubuntu
Platform | andriod

### Issue Description
I followed the android case,But here's the problem

Build command failed.
Error while executing process /home/yingzm/Android/Sdk/cmake/3.6.4111459/bin/cmake with arguments {-H/home/yingzm/AndroidStudioProjects/realsense_app/app -B/home/yingzm/AndroidStudioProjects/realsense_app/app/.externalNativeBuild/cmake/debug/armeabi-v7a -DANDROID_ABI=armeabi-v7a -DANDROID_PLATFORM=android-19 -DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/home/yingzm/AndroidStudioProjects/realsense_app/app/build/intermediates/cmake/debug/obj/armeabi-v7a -DCMAKE_BUILD_TYPE=Debug -DANDROID_NDK=/home/yingzm/Android/Sdk/ndk-bundle -DCMAKE_CXX_FLAGS=-std=c++11 -fexceptions -DCMAKE_TOOLCHAIN_FILE=/home/yingzm/Android/Sdk/ndk-bundle/build/cmake/android.toolchain.cmake -DCMAKE_MAKE_PROGRAM=/home/yingzm/Android/Sdk/cmake/3.6.4111459/bin/ninja -GAndroid Gradle - Ninja}

CMake Error at CMakeLists.txt:14 (add_subdirectory):
    /home/yingzm/AndroidStudioProjects/realsense_app/app/src/main/cpp/librealsense
  does not contain a CMakeLists.txt file.
-- Configuring incomplete, errors occurred!

RealSense-Customer-Engineering, 2018-06-27:
[Realsense Customer Engineering Team Comment]
@yingthink Seems to be the similar issue with https://intel.zendesk.com/agent/tickets/2759

Please use the CMakeLists.txt in the link below.

https://github.com/IntelRealSense/librealsense/blob/v2.13.0/doc/android/CMakeLists.txt_

RealSense-Customer-Engineering, 2018-07-05:
[Realsense Customer Engineering Team Comment]
@yingthink Is the issue resolved?

RealSense-Customer-Engineering, 2018-07-18:
[Realsense Customer Engineering Team Comment]
@yingthink  Is the issue resolved? If so, we'll close the ticket accordingly.

RealSense-Customer-Engineering, 2018-07-26:
[Realsense Customer Engineering Team Comment]
@yingthink Any update about the issue? Can you build successfully now?

yingthink, 2018-07-30:
@RealSense-Customer-Engineering 
issue was settled

closed 2018-07-30 by yingthink