realsenseai/librealsense #1893: Build an Android Application Error

aliray, 2018-06-15:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10)  | 
| Platform                                 | PC  |
| SDK Version                            |  2.0   | 
| Language                            |java | 

### Issue Description
try to follow the "Build an Android Application for RealSense SDK" but below error exists!!!!!

Build command failed.
Error while executing process E:\android_sdk\cmake\3.6.4111459\bin\cmake.exe with arguments {-HE:\android_space\realsense_app\app -BE:\android_space\realsense_app\app\.externalNativeBuild\cmake\debug\armeabi-v7a -DANDROID_ABI=armeabi-v7a -DANDROID_PLATFORM=android-19 -DCMAKE_LIBRARY_OUTPUT_DIRECTORY=E:\android_space\realsense_app\app\build\intermediates\cmake\debug\obj\armeabi-v7a -DCMAKE_BUILD_TYPE=Debug -DANDROID_NDK=E:\android_sdk\ndk-bundle -DCMAKE_CXX_FLAGS=-std=c++11 -fexceptions -DCMAKE_TOOLCHAIN_FILE=E:\android_sdk\ndk-bundle\build\cmake\android.toolchain.cmake -DCMAKE_MAKE_PROGRAM=E:\android_sdk\cmake\3.6.4111459\bin\ninja.exe -GAndroid Gradle - Ninja}
-- Check for working C compiler: E:/android_sdk/ndk-bundle/toolchains/llvm/prebuilt/windows-x86_64/bin/clang.exe
-- Check for working C compiler: E:/android_sdk/ndk-bundle/toolchains/llvm/prebuilt/windows-x86_64/bin/clang.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: E:/android_sdk/ndk-bundle/toolchains/llvm/prebuilt/windows-x86_64/bin/clang++.exe
-- Check for working CXX compiler: E:/android_sdk/ndk-bundle/toolchains/llvm/prebuilt/windows-x86_64/bin/clang++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at src/main/cpp/librealsense2/CMakeLists.txt:22 (file):
CMake Error at src/main/cpp/librealsense2/CMakeLists.txt:22 (file):
CMake Error at src/main/cpp/librealsense2/CMakeLists.txt:22 (file):
-- Info: REALSENSE_VERSION_STRING=-1.-1.-1
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Prepare RealSense SDK for Android OS (armeabi-v7a)
CMake Warning at src/main/cpp/librealsense2/CMakeLists.txt:576 (message):
CMake Error at src/main/cpp/librealsense2/CMakeLists.txt:637 (add_subdirectory):
CMake Error at src/main/cpp/librealsense2/CMakeLists.txt:643 (add_subdirectory):
CMake Error at src/main/cpp/librealsense2/CMakeLists.txt:762 (include):
    E:/android_space/realsense_app/app/src/main/cpp/librealsense2/third-party/realsense-file/config.cmake
CMake Error: File E:/android_space/realsense_app/app/src/main/cpp/librealsense2/CMake/realsense2Config.cmake.in does not exist.
CMake Error: File E:/android_space/realsense_app/app/src/main/cpp/librealsense2/cmake_uninstall.cmake does not exist.
CMake Error at src/main/cpp/librealsense2/CMakeLists.txt:876 (configure_file):
CMake Error: File E:/android_space/realsense_app/app/src/main/cpp/librealsense2/config/librealsense.pc.in does not exist.
CMake Error at src/main/cpp/librealsense2/CMakeLists.txt:888 (configure_file):
-- Configuring incomplete, errors occurred!
See also "E:/android_space/realsense_app/app/.externalNativeBuild/cmake/debug/armeabi-v7a/CMakeFiles/CMakeOutput.log".
Build command failed.
Error while executing process E:\android_sdk\cmake\3.6.4111459\bin\cmake.exe with arguments {-HE:\android_space\realsense_app\app -BE:\android_space\realsense_app\app\.externalNativeBuild\cmake\release\armeabi-v7a -DANDROID_ABI=armeabi-v7a -DANDROID_PLATFORM=android-19 -DCMAKE_LIBRARY_OUTPUT_DIRECTORY=E:\android_space\realsense_app\app\build\intermediates\cmake\release\obj\armeabi-v7a -DCMAKE_BUILD_TYPE=Release -DANDROID_NDK=E:\android_sdk\ndk-bundle -DCMAKE_CXX_FLAGS=-std=c++11 -fexceptions -DCMAKE_TOOLCHAIN_FILE=E:\android_sdk\ndk-bundle\build\cmake\android.toolchain.cmake -DCMAKE_MAKE_PROGRAM=E:\android_sdk\cmake\3.6.4111459\bin\ninja.exe -GAndroid Gradle - Ninja}
-- Check for working C compiler: E:/android_sdk/ndk-bundle/toolchains/llvm/prebuilt/windows-x86_64/bin/clang.exe
-- Check for working C compiler: E:/android_sdk/ndk-bundle/toolchains/llvm/prebuilt/windows-x86_64/bin/clang.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: E:/android_sdk/ndk-bundle/toolchains/llvm/prebuilt/windows-x86_64/bin/clang++.exe
-- Check for working CXX compiler: E:/android_sdk/ndk-bundle/toolchains/llvm/prebuilt/windows-x86_64/bin/clang++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at src/main/cpp/librealsense2/CMakeLists.txt:22 (file):
CMake Error at src/main/cpp/librealsense2/CMakeLists.txt:22 (file):
CMake Error at src/main/cpp/librealsense2/CMakeLists.txt:22 (file):
-- Info: REALSENSE_VERSION_STRING=-1.-1.-1
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Prepare RealSense SDK for Android OS (armeabi-v7a)
CMake Warning at src/main/cpp/librealsense2/CMakeLists.txt:576 (message):
CMake Error at src/main/cpp/librealsense2/CMakeLists.txt:637 (add_subdirectory):
CMake Error at src/main/cpp/librealsense2/CMakeLists.txt:643 (add_subdirectory):
CMake Error at src/main/cpp/librealsense2/CMakeLists.txt:762 (include):
    E:/android_space/realsense_app/app/src/main/cpp/librealsense2/third-party/realsense-file/config.cmake
CMake Error: File E:/android_space/realsense_app/app/src/main/cpp/librealsense2/CMake/realsense2Config.cmake.in does not exist.
CMake Error: File E:/android_space/realsense_app/app/src/main/cpp/librealsense2/cmake_uninstall.cmake does not exist.
CMake Error at src/main/cpp/librealsense2/CMakeLists.txt:876 (configure_file):
CMake Error: File E:/android_space/realsense_app/app/src/main/cpp/librealsense2/config/librealsense.pc.in does not exist.
CMake Error at src/main/cpp/librealsense2/CMakeLists.txt:888 (configure_file):
-- Configuring incomplete, errors occurred!
See also "E:/android_space/realsense_app/app/.externalNativeBuild/cmake/release/armeabi-v7a/CMakeFiles/CMakeOutput.log".

RealSense-Customer-Engineering, 2018-06-27:
[Realsense Customer Engineering Team Comment]
@aliray  Please refer to pull request as below to try again.

https://github.com/IntelRealSense/librealsense/pull/1853

RealSense-Customer-Engineering, 2018-07-05:
[Realsense Customer Engineering Team Comment]
@aliray Is there any update? Is the issue resolved?

RealSense-Customer-Engineering, 2018-07-18:
[Realsense Customer Engineering Team Comment]
@aliray Do you still have this problem? Please update. If resolved, we'll close the ticket in 5 days.

RealSense-Customer-Engineering, 2018-07-30:
[Realsense Customer Engineering Team Comment]
@aliray Didn't get update from you. We'll close this ticket. Please create another new ticket if you still have questions or problems.

closed 2018-07-30 by RealSense-Customer-Engineering