realsenseai/librealsense #1894: Request enhancements: rs-convert wants 3d distance csv support

longjj, 2018-06-15:
### Issue Description

`rs-convert` in SDK version 2.12.0 is awesome!

I wonder whether it will support to convert each ROS-bag file to a format (maybe csv) that contains the real 3d distances info of each pixel in the depth matrix? #1868

It seems to use `rs2_deproject_pixel_to_point(...)` to get the distances in x and y axis like z axis using `get_distance()` or something like that.

It is my first time to new an issue for enhancements, and I will give reply if my statement makes you confused.

Thx.

dorodnic, 2018-06-15:
Hi @longjj, thanks for the feedback. 
What would be the exact content of the CSV you would like to be added?

dorodnic, 2018-06-16:
We can add this in a future release. 
Please note that what you are asking is very similar to the content of PLY exporter, the only difference being that PLY format we use is binary and not textual. However, it is essentially a binary dump of (x, y, z) coordinates.

longjj, 2018-06-16:
@dorodnic Thank you so much!

I used matlab to read the PLY format file. It's so cool ! 

[image]

Enlarge the red region and I can see my face!

[image]

I will try to use PLY file first, while I consider the textual representation of (x, y, z) is **more intuitive and more concise** if we don't care the rgb of an exact point.

In addition, should I close the issue now?  I am not sure since it is my first time to new an issue for enhancements.

dorodnic, 2018-06-16:
Keeping the issue open will help our team to track and integrate it into our internal backlog.

RealSenseCustomerSupport, 2022-04-23:
----------------------------------------------

Hello @longjj  My name is Marty Grover and I will be handling your case.

Could you let me know whether you still require assistance with this case, please?

Thanks very much,

Marty Grover
Intel RealSense Support Engineer

MartyG-RealSense, 2022-04-23:
**Update:** the Intel feature request to export 3D depth information in a csv that was opened in June 2018 is still active but has not had progress for 2 years at the time of writing this as it is considered to be a low-priority feature.

MartyG-RealSense, 2026-10-25:
Case closed due to the significant amount of time that has passed with no developments.

closed 2026-10-25 by MartyG-RealSense