realsenseai/librealsense #1896: D415, Unity, Pointcloud, latency in aligner process

PaulBre, 2018-06-15:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.09.11.00 | 
| Operating System & Version | Win 10  | 
| Platform                                 | PC |
| SDK Version                            |  2.11.1 (development) | 
| Language                            |  unity   | 

### Issue Description
In the Unity RealSensePointCloud scene in I notice severe latency when I use a D415 camera. When I use a SR300 camera the latency is hardly noticeably. When I add a framecounter in the PointCloudGenerator OnFrames method and use a D415 camera at 30Hz the camera frames arrive at about 30 Hz as long as the camera image is not changed (i.e. as long as objects in front of a fixed camera does not move). As long as objects are moving (i.e. arm waving) the rate of the arriving camera frames drops dramatic. Can anybody explain or fix this "dynamic" latency effect? Why only on D415 and not on SR300?

When I skip the align function in Onframes (see PointCloudGenerator.cs code below) the latency on the D415 disappears! Maybe this helps debugging.

    int framecnt = 0;
    private void OnFrames(FrameSet frames)
    {
        //using (var aligned = aligner.Process(frames))
        //{
            //using (var colorFrame = aligned.ColorFrame)
            //using (var depthFrame = aligned.DepthFrame)
            using (var colorFrame = frames.ColorFrame)
            using (var depthFrame = frames.DepthFrame)
            {
                if (depthFrame == null)
                {
                    Debug.Log("No depth frame in frameset, can't create point cloud");
                    return;
                }

                if (!UpdateParticleParams(depthFrame.Width, depthFrame.Height))
                {
                    Debug.Log("Unable to craete point cloud");
                    return;
                }

                using (var points = pc.Calculate(depthFrame))
                {
                    setParticals(points, colorFrame);
                }
            }
        //}
        framecnt++;
    }

dorodnic, 2018-06-15:
Hi @PaulBre 
My guess for the difference between two cameras is that SR300 is capped at VGA, while default D415 depth is HD, requiring more processing. 
We need to review the Unity layer to make sure there are no large queues anywhere, and we are also improving the optimization of the aligner to help with this issue.

PaulBre, 2018-06-15:
Hi @dorodnic ,
Thanks for your quick response.
For both SR300 and D415 I use the same example Unity scene where the RealSenseDevice is initialized with Depth Z16 640x480 and Infrared Y8 640x480 and Color Rgb8 640x480.
What I really don't understand is why the latency is depending on movement in the field of view of the camera.

CodaxD, 2018-06-15:
I ran into something similar with the D435, it seemed to be linked to changing Exposure settings.  Manually adjusting the Exposure settings seemed to cause frame loss (Unity FPS was still high) especially on the color stream.  However, the problem could occasionally be resolved by readjusting the exposure or turning on AutoExposure.

It is reproducible in the  RealsenseTextures.unity if you add a RealsenseDeviceInspector component to the RealsenseDevice gameobject.  Press play, then turn off "AutoExposure" on the color stream.  Next make a change to the exposure value.  I've been able to reproduce frame loss 7 / 10 times I performed these steps.

I have seen frame loss by setting all the streams to max resolution (1280x720@30 for Infrared/Depth and 1920x1080@30fps) even with Auto Exposure

RealSense-Customer-Engineering, 2018-06-29:
[Realsense Customer Engineering Team Comment]
Hi @PaulBre,

The frame latency could also be due to the exposure change based on the test scene as @CodaxD mentioned. How it works if you try to fix IR exposure value instead of AE for D415?

PaulBre, 2018-07-17:
As proposed above I tested the RealSensePointCloud.unity scene on latency (RealsenseDeviceInspector added). I also added a GUI to monitor both the Unity fps and the "camera fps" (Process Mode Multithread). In the default situation (AutoExposure ON) the "camera fps" drops dramatic when large objects are moved in and out of the camera field of view (especially with a black coat) while the Unity fps stays at about the same value. After switching the EnableAutoExposure of the Stereo Module ON => OFF the latency effect disappears! Switching the EnableAutoExposure of the RGB camera has no effect on latency. Maybe this helps pinpointing this latency issue. I consider switching off the Auto Exposure a work-around. Note that this test has been performed in SDK version 2.11.1 (development).

RealSense-Customer-Engineering, 2018-07-26:
[Realsense Customer Engineering Team Comment]

Hi @PaulBre,

Please check librealsense 2.14.0 with mesh-based solution in Unity to see how it improves the performance.

PaulBre, 2018-07-27:
Librealsense 2.14.0 fixes this latency issue. For pointcloud use in Unity (with aligned color image "mapped on the depth data") 2.14.0 is a great improvement! Issue closed.

closed 2018-07-27 by PaulBre