realsenseai/librealsense #1897: which kind of intel@realsense supports slam?

xufeifeiWHU, 2018-06-15:
Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 } | 

which kind of intel@realsense supports slam?
only I can find the document about slam on Intel sites is published on year 2016 ,
So i am here to get more new information about realsense and slam
Thanks for your time and look for your reply

RealSense-Customer-Engineering, 2018-06-18:
[Realsense Customer Engineering Team Comment]
hi @xufeifeiWHU,

would like to know if you have NDA with Intel? if so please raise this question through sales channel or Zendesk system, thanks.

RealSense-Customer-Engineering, 2018-06-26:
[Realsense Customer Engineering Team Comment]
hi @xufeifeiWHU,

would like to confirm if you reached our sales channel already?
thanks.

closed 2018-07-01 by xufeifeiWHU