realsenseai/librealsense #1899: Memory leaks

sennevanoppen, 2018-06-15:
| Required Info | |
|-|-|
| Camera Model | D400
| Firmware Version | 05.09.11.00
| Operating System & Version | Win 10
| Kernel Version (Linux Only) | na
| Platform | PC
| SDK Version | 2.11.1
| Language | C#

### Issue Description
The Device class (and more) doesn't contain a destructor or dispose.

Calling QueryDevices continuously creates new Device objects which live on after. When using this method numerous times it ultimately causes a OutOfMemoryException.

RealSense-Customer-Engineering, 2018-06-20:
[Realsense Customer Engineering Team Comment]
@sennevanoppen
I did find the memory leak of this function, and will let engineer knows about this issue. Can you let me know why you need to query device often? To check if device exist?

dorodnic, 2018-08-27:
Should be resolved on the development branch and going to be part of 2.16 release

dorodnic, 2018-09-25:
I believe this is resolved now, please open a new issue if you encounter any problem

closed 2018-09-25 by dorodnic