realsenseai/librealsense #1900: Installation Failure : patching file drivers/media/usb/uvc/uvcvideo.h Hunk #1 FAILED at 164.

angeltop, 2018-06-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 / D400 | 
| Firmware Version                   |  | 
| Operating System & Version |   Linux (Ubuntu 16.04)   | 
| Kernel Version (Linux Only)    |  4.13.0-45                                      | 
| Platform                                 | PC.  |
| SDK Version                            |   2                          | 
| Language                            |  C                           | 

### Issue Description
After  following the installation instructions found at https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md
for my Ubuntu 16.04 PC it failed to patch the uvcvideo module.

The first that I could not do was the specific to Ubuntu 16 command:

> Ubuntu 16:
> sudo apt-get install cmake3 libglfw3-dev

It  could not find cmake3, I have cmake I hope that is good enough.

And then it failed in the command:
> ./scripts/patch-realsense-ubuntu-lts.sh

Output:

Package required libusb-1.0-0-dev:  - found
Package required libssl-dev:  - found
Cloning into './ubuntu-xenial-hwe'...
remote: Counting objects: 67939, done.
remote: Compressing objects: 100% (62201/62201), done.
remote: Total 67939 (delta 7416), reused 18193 (delta 4648)
Receiving objects: 100% (67939/67939), 179.02 MiB | 3.59 MiB/s, done.
Resolving deltas: 100% (7416/7416), done.
Checking connectivity... done.
Checking out files: 100% (64412/64412), done.
Applying realsense-uvc patch
patching file drivers/media/usb/uvc/uvc_driver.c
patching file drivers/media/usb/uvc/uvcvideo.h
Hunk #1 succeeded at 153 (offset 1 line).
patching file drivers/media/v4l2-core/v4l2-ioctl.c
Hunk #1 succeeded at 1246 (offset 7 lines).
patching file include/uapi/linux/videodev2.h
Hunk #1 succeeded at 668 (offset 6 lines).
Applying realsense-metadata patch
patching file drivers/media/usb/uvc/uvc_driver.c
Hunk #1 succeeded at 2771 (offset 29 lines).
patching file drivers/media/usb/uvc/uvc_video.c
Hunk #1 succeeded at 1226 (offset 4 lines).
Hunk #2 succeeded at 1264 (offset 4 lines).
patching file drivers/media/usb/uvc/uvcvideo.h
Hunk #1 FAILED at 164.
Hunk #2 succeeded at 208 (offset 23 lines).
1 out of 2 hunks FAILED -- saving rejects to file drivers/media/usb/uvc/uvcvideo.h.rej

Should I use an older commit?

abecciu, 2018-06-19:
Ubuntu 16.04 recently updated the version of the HWE kernel to 4.15. That's why the patch fails. I manually downloaded the sources for the latest 4.13 HWE and manually applied the patches. Then I modified the script to skip downloading kernel sources and applying patches.

RealSense-Customer-Engineering, 2018-06-19:
[Realsense Customer Engineering Team Comment]
thanks for your help @abecciu.

hi @angeltop,
would like to know if abecciu's suggestion solve your issue?

TonyFaith, 2018-06-22:
@abecciu would you please give more details about "downloaded the sources for the latest 4.13 HWE and manually applied the patches" and  "modified the script to skip downloading kernel sources and applying patches" , thanks a lot!

ev-mp, 2018-06-26:
@angeltop  ,@TonyFaith,
The patches for Ubuntu HWE with 4.13 have been updated in the latest version available on github.
See the PR link posted above for details.

Please give it a try and update.

angeltop, 2018-06-28:
@ev-mp thank you for the update. I was able to patch uvcvideo :)

closed 2018-06-28 by angeltop