realsenseai/librealsense #1902: Preset for dense SLAM

martinakos, 2018-06-18:
I'm trying to use a D415 for a dense SLAM system. (For example KinnectFusion, or ElasticFusion, etc). These SLAM systems were designed and tested with a Primesense/Kinnect type of camera (structured light).  When I test these SLAM systems with a D415 I get very poor performance in comparison to Primesense.  I seem to get very sparse depth maps and if I use the HighDensity Preset I get too much noise. 

After playing for a couple of days with the presets found [here](https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets) and  the other settings that I can see in the realsense-viewer tool I improved a bit the depth map adjusting the following settings:
Starting from Default preset.
Disabling autoexposure: seems to make everything a bit less noisy.
Depth Control/DS second Peak th: high seems to make the scene less noisy at short ranges.
SLO Penalty Control/K2 penalty:  high value seems to make scene a bit denser.
Depth clamp: 800-6000 

However, it's still more sparse and noisy that with a primesense depth map, and the SLAM system doesn't work so well, the maps are substantially more sparse. 

I wonder if you could produce a preset for the D415 optimised for dense SLAM.

RealSense-Customer-Engineering, 2018-06-21:
[Realsense Customer Engineering Team Comment]
@martinakos
Sorry that we don't have the SLAM preset, but there are documents guide you how to tune up the depth image per your usage, please reference to the documents attached.
https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/BKMs_Tuning_RealSense_D4xx_Cam.pdf
https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Intel-RealSense-Depth-PostProcess.pdf

martinakos, 2018-06-21:
I read these documents before and I have spent a few days tuning parameters. I guess I can try more on the side of post-processing.  It would be great though if you could create that SLAM preset.

RealSense-Customer-Engineering, 2018-06-22:
[Realsense Customer Engineering Team Comment]
@martinakos
Thank you for the feedback, I will discuss internally if it may be provided in the future, but currently there is no plan for providing SLAM preset. I close the issue first, if you have any related question, feel free to create new ticket or reopen it.

closed 2018-06-22 by RealSense-Customer-Engineering

agrunnet, 2018-06-22:
“High Accuracy” setting is currently our best recommendation for this but some people have had success with “high density” as well.

martinakos, 2018-06-26:
I tested the realsense D415 with dense SLAM systems such as KinectFusion or ElasticFusion, and using the "High Accuracy" the depth map was too sparse for the SLAM system to work well, and with "High Density" the depth map was too noisy for the SLAM system to work well.