realsenseai/librealsense #1903: Windows Precompiled Binaries are not up to date

Chimdi511, 2018-06-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C}                          | 

With every new source version of librealsense, the precompiled binaries are left untouched. Why should the binary installer be different from the source version of the library ?

RealSense-Customer-Engineering, 2018-06-19:
[Realsense Customer Engineering Team Comment]
hi @Chimdi511,

sorry i don't get you, do you mean tools or examples from **Intel.RealSense.SDK.exe of v2.12.0, are not the same from Source code (zip) of v2.12.0?**

Chimdi511, 2018-06-21:
I mean the tools and examples from Intel.RealSense.SDK.exe. For example, the pcl examples aren't in the execeutable.

Julz-s, 2018-06-25:
Same for the rs-convert tool.
I had to download the source and compile everything to get this tool :)

dorodnic, 2018-06-25:
Hi @Chimdi511, @Julz-s 

We will gradually keep expanding Windows / Linux and PiPy installers, but all new features are first introduced and tested in the open-source environment and only later integrated into the different packages (so the newly added `rs-convert` should be added to the installer soon). 

Another consideration is what dependencies we want to redistribute. I don't think PCL examples will ever be part of the installer, since PCL itself has no stable ABI (we would have to redistribute a separate binary for each version of Visual Studio)

closed 2018-07-23 by RealSense-Customer-Engineering