realsenseai/librealsense #1905: How to change depth units in python

Litvak1, 2018-06-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |   Linux (Ubuntu 16.04)  | 
| Kernel Version (Linux Only)    |  4.13.0-45-generic                                        | 
| Platform                                 | PC |
| SDK Version                            | ? | 
| Language                            |  python                       | 

Hi,

As explained [here](https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0#point-coordinates), I am trying to change the depth units from millimeter (depth_units  = 1000) to a tenth of that (depth_units  = 100).

I couldn't find how to use 'rs2_set_option' in python.

Could you please send some example?

Also, will the minimum distance of depth images be affected by this change?

Thank you!

0xLeon, 2018-06-18:
Options are set on a sensor level. Therefore, options are set on sensor objects. The procedure is outlined in a script I wrote: https://github.com/0xLeon/StereoImageProcessing/blob/master/Lab/GetDepthFrames.py#L55
Instead of `rs.option.emitter_enabled` you would set `rs.option.depth_units`. Also, keep in mind, depth units are always specified in meter. The depth unit option controls the meaning of one unit in the depth stream data format.

Litvak1, 2018-06-19:
Hi @0xLeon.

Thanks for answering my question.

I've tried what you said and got an error.

This is the code:

> profile = pipeline.start()
> depth_sensor = profile.get_device().first_depth_sensor()
> if depth_sensor.supports(rs.option.depth_units):
>     depth_sensor.set_option(rs.option.depth_units, 100.0)
> 

And this is the error:
`RuntimeError: OpCodes do not match! Sent 43 but received -6!
`

If I try to change the depth units in realsense-viewer it works.

Do you have some suggestion?

0xLeon, 2018-06-19:
I think the example is not up-to-date (can't say if it was every true, I'm not that long into RealSense development). If you have a look at `depth_sensor.get_option(rs.option.depth_units)`, you can see the returned value is a float with a default value of around `0.001`. This corresponds to the default depth unit of 1 mm expressed in meters. So if you want to have depth unit of 0.1 mm, you would pass this in meters to the call, which would be 0.0001: `depth_sensor.set_option(rs.option.depth_units, 0.0001)`.

I'm not completely sure if this is actually true, I didn't dive into the C++ code so far, only worked with the higher level Python bindings. And I haven't checked on the outcome of changing `rs.option.depth_units` so far. Maybe you can have a look at this, I'd be happy to read about your results.

Litvak1, 2018-06-19:
It worked!

Exactly as you described it.

Thanks a lot @0xLeon !

If you have the time, please see if you could help me with another issue I submitted -  **[Transforming pixel from a depth image to world coordinates](https://github.com/IntelRealSense/librealsense/issues/1904)**

closed 2018-06-19 by Litvak1