realsenseai/librealsense #1907: Python bindings only worked after manual installation of VC++ 2017 Redistributables

0xLeon, 2018-06-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.09.11.00 | 
| Operating System & Version |   Win10 x64  | 
| Kernel Version (Linux Only)    |  -                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.12.0                           | 
| Language                            |  Python                          | 

### Issue Description
Just now, I installed the SDK from the Installer. Directly after installation, I wasn't able to load the module from Python with the following error message: `ImportError: DLL load failed: Das angegebene Modul wurde nicht gefunden.`. I tracked this down to be a VC++ 2017 Redistributable problem probably. I installed the amd64 package of those, restarted my machine and it worked flawless.

Now I'm not sure if this issue would have persisted without the manual installation of the VC++ 2017 Redistributables. Does the SDK installer ship these? If yes, you should probably suggest a restart after installation as well. Though the Viewer worked without a restart, the Python library did not. I did build the library before on my main machine (now using a temporary replacement due to hardware failure on my main machine, that's why Im using the pre-copiled binaries). Those worked out-of-the-box probably due to the necessary runtime librares already being available on the machine.

RealSense-Customer-Engineering, 2018-06-21:
[Realsense Customer Engineering Team Comment]
Hi @0xLeon
Thanks for this input, myself also found the vc++ redist package existence  matter
we will look into this issue then update you any progress

dorodnic, 2018-07-17:
We believe this should resolved in [2.14.0 SDK](https://github.com/IntelRealSense/librealsense/releases/download/2.14.0/Intel.RealSense.Viewer.exe), could you please confirm?

RealSense-Customer-Engineering, 2018-07-20:
[Realsense Customer Engineering Team Comment]
@dorodnic
Yes, the library provided in 2.14.0 SDK can fix the report issue
that user no need to install VC++ 2015 or 2017 redistributable in order to run the python sample script

Hi @0xLeon
thanks for your input and suggestion
We had released newer version of SDK which has the precompiled library which should be able to fixed the reported issue
please let us know if there any problem.

0xLeon, 2018-07-20:
Currently I do not have a clean Windows image without any VC++ Redistributables installed to check if the issue is fixed, sorry. Someone could quickly set up a Win10 VM and check if the Python wrapper works out of the box after installing the SDK via the Windows installer. Currently, I'm busy with work and other projects.

dorodnic, 2018-07-26:
Please open new issue if you encounter something similar again. We tested on a clean Win 10 and seem to Ok now.

closed 2018-07-26 by dorodnic