realsenseai/librealsense #1908: Debian repo broken?

floe, 2018-06-19:
It seems that the repository described in the Linux installation instructions is currently broken:
```
E: Failed to fetch http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/xenial/InRelease  403  Forbidden [IP: 52.218.48.217 80]
```
Same error (403) also happens with https, and other release names.

RealSense-Customer-Engineering, 2018-06-20:
[Realsense Customer Engineering Team Comment]
Hello and thank you for your inquiry,

We tested the Linux installation method depicted here:  https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md, and did not see any issues downloading and installing packages.

>  If you have not registered the Amazon S3 server key, you will receive the forbidden IP error.  See attached screenshot for complete Linux instructions so that you can verify every command runs successfully, especially highlighted command, if ran successfully (on a network without firewalls or external network altogether) should fix issue.

Thank you.

floe, 2018-06-21:
Just to be sure, I removed and re-added the repo by copy-pasting the instructions again, and I am still getting the same error. I am located in Germany, is it possible that this is due to some regional configuration?

RealSense-Customer-Engineering, 2018-06-21:
[Realsense Customer Engineering Team Comment]
Thank you for the information floe,

The repositories are working as intended, we were able to replicate your issue within a virtual machine environment that has limited network features.
This solution should work for your specific situation.
Please run the following command and Amazon S3 repositories will be updated as necessary, enter commands exactly as shown:
sudo apt -o Acquire::AllowInsecureRepositories=true \
> -o Acquire::AllowDowngradeToInsecureRepositories=true \
> update [ENTER]

Please let us know if you have any further issues.

floe, 2018-06-22:
OK, I think the issue actually appeared because I've updated to Bionic (18.04), and the newest version of apt will now always request an `InRelease` file first, which is supposed to contain inline signatures (see also https://wiki.debian.org/DebianRepository/Format). The long-term fix will consequently be to add an `InRelease` file to the repo, using `AllowInsecureRepositories` is only a workaround which may actually cause security issues if used constantly.

floe, 2018-06-22:
P.S. as indicated here (http://wiki.blue-panel.com/index.php/Create_Signed_APT_repository), creating `InRelease` just requires one extra gpg command. Shouldn't be too difficult to implement.

RealSense-Customer-Engineering, 2018-06-22:
[Realsense Customer Engineering Team Comment]
floe,

Thank you very much for bringing this to our attention.  I will forward this info to dev team to implement fix in future releases.

closed 2018-06-22 by RealSense-Customer-Engineering