realsenseai/librealsense #1912: Over-exposure Artifacts in Depth

xtrawurst, 2018-06-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 5.9.11 | 
| Operating System & Version |   Win10 | 
| Platform                                 | PC |
| SDK Version                            |  2.11.1 | 

### Issue Description
We are currently working with the RS in depth and trying to understand best settings / best practices.

One thing we are running into is artifacts due to over-exposure in the IR image. Below video shows the effect.

Of course, over-exposure is a thing to avoid, but so is under-exposure (as it does come with poor density and higher noise in the depth, as pointed out in the white paper). Hence we'd generally prefer an over-exposed input over an under-exposed input -- if there was a way to get rid of the artifacts.

In another video below I show how the default-settings-AE (which tends to under-expose) performs against manual exposure. You can see a clearly elevated noise level and a sparser depth map with the under-exposing AE vs. the ME. 

The artifacts appear within the boundaries of clearly over-exposed areas. It seems that the depth engine is "hallucinating" fill-in depth values where there's really no signal. The result of the artifacts are "bellys" and "dents" in the depth. Very visible on flat walls.

**We wonder if there's a way to tell the DS5 that it should rather ignore these pixels with no signal, and not assign a depth value, as opposed to fill them in with bad values.**

The K1 and K2 thresholds seem to help with this, but they come at the expense of sparsity in the rest of the depth image. Even more so for the "high accuracy" preset.

Given that the over-exposure artifacts seem to be limited to areas in the IR image that are literally white and 0-gradient, I think that a fast host-sided filter could be implemented quickly within librealsense. Would this be feasible?

 [Video 1 (over-exposure artifacts)](https://vimeo.com/275823145)

 [Video 2 (AE vs ME)](https://vimeo.com/275823870)

ev-mp, 2018-06-21:
Hello @xtrawurst ,
Implementing a host-side filter only cannot affect the artifact directly as the depth is calculated on ASIC.
Librealsense provides "Depth exposure ROI" control that instructs the depth sensor which portion of the Field of View will be used for auto-exposure adjustments.

I guess one way to address this question would probably comprise of implementing a host-side filter to identify the over-exposed zones, and then combine it with the above control to implement a control feedback loop:
1- Host-side filter will calculate the sub-frame that has less/no overexposure.
2. the Resulted region will be sent back to device to adjust Auto-exposure ROI.

This is actually how a Fisheye sensor autoexposure feature was implemented (on host) in librealsense1 for ZR300 depth camera.

xtrawurst, 2018-06-21:
@ev-mp I'm suggesting something simpler than that: Detect white areas of 0-gradient in the IR image and set to 0 all corresponding pixels in the depth image.

Would that be something that is quick to integrate into librealsense?

Julz-s, 2018-06-25:
Well, I've not tried this but you could use some sobel or scharr like filter on one of the IR images and threshold the results to get areas with small gradient.
Implementing such filters should be quite straightforward.

xtrawurst, 2018-06-25:
@Julz-s Could certainly do this but to me it seems like such an obvious improvement to the RealSense product that my hope was that RS engineering could take this on.

RealSense-Customer-Engineering, 2018-07-03:
[Realsense Customer Engineering Team Comment]
Hi @xtrawurst,

As @ev-mp mentioned, a host-side filter can be implement to feedback to adjust auto-exposure ROI. In addition, IR emitter power can be also adjusted accordingly to avoid over exposure. Still, need to further discuss this feature request in host-side filter internally. By the way, do you already implement anything for this?

dorodnic, 2018-07-11:
Hi @xtrawurst @Julz-s 
We had an internal discussion on the topic with relevant algo stakeholders and decided not to implement this for now - having over / under saturated IR image is itself an issue and needs to be addressed via camera controls (exposure, gain and exposure ROI). 
If you decide to implement such filter for your project, we would be happy to integrate a pull-request. 
Thank you for the feedback.

closed 2018-07-11 by dorodnic