realsenseai/librealsense #1914: The requested pixel format 'INZI' is not natively supported by the Linux kernel and likely requires a patch!

summer1216, 2018-06-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Operating System & Version |   UBUNTU14 | 
| Kernel Version (Linux Only)    |  Linux 4.4.0-128-generic                                        | 
| Platform                                 | PC |

I updated my kernel to Linux 4.4.0-128-generic  when I was manually installing the librealsense, and after that it was working well for some months. 

But after some days not working with the camera, today I work again with the camera, I suddenly got the following error which was not there before. 

_```
terminate called after throwing an instance of 'rs2::backend_error'
  what():  The requested pixel format 'INZI' is not natively supported by the Linux kernel and likely requires a patch!
Alternatively please upgrade to kernel 4.12 or later. Last Error: Inappropriate ioctl for device
```_

I checked the the two related issues and did not fix my problem.

ev-mp, 2018-06-19:
Hi @summer1216 , Is it possible that you had run Ubuntu update? 

You can add kernel log when connecting the camera, but if you can see complains about missing formats, especially INZI then the patched uvcvideo module has been replaced either by kernel up/down-grade or manually.

Rerun the kernel patching part of the Linux installation manual and update.

summer1216, 2018-06-20:
@ev-mp  thank you for your reply. It seems that maybe I have updated the kernel from Linux 4.4.0-127-generic to Linux 4.4.0-128-generic, which I didn't mean to. :( 

I tried to rerun the patching part, the following error occurs, and camera can not be detected anymore. 

`Applying the patched module ... modprobe: ERROR: could not insert 'uvcvideo': Invalid argument
Failed to insert the patched module. Operation is aborted, the original module is restored
Verify that the current kernel version is aligned to the patched module version
modprobe: ERROR: could not insert 'uvcvideo': Invalid argument
`

ev-mp, 2018-06-20:
Make sure to perform the linux preps stage: `sudo apt update && ....` before running the patch.
Additionally, can you post the kernel log when the error occurs (`dmesg`) ?

ev-mp, 2018-06-20:
I would suggest to temporally revert to kernel 4.4.0-127 unless you have explicit requirements, since:
1. There is a good chance that the previous setup with patched kernel module is still intact.
2. Even if not - the kernel version would be identical to the sources downloaded from Ubuntu, so patching the kernel should succeed.

We will try reproduce this situation with kernel 4.4.0-128 in the meanwhile and update  accordingly

summer1216, 2018-06-21:
@ev-mp  Thank you for your suggestion. I reinstalled Kernel 4.4.0-127 and reboot with it, after redoing patching, it works fine now. I still don't know why after removing the old kernel completely, patching still recognizes I have the old kernel. But so far I can work with this. Thanks a lot for your help!

TonyFaith, 2018-06-22:
@ev-mp @summer1216  thanks a lot! I meets the same problems as @summer1216 . Would you please give some instruction about reinstall  Kernel 4.4.0-127 ?? Thanks.

RealSense-Customer-Engineering, 2018-06-24:
[Realsense Customer Engineering Team Comment]
Hi @summer1216,

Still need any further help? If your issue resolved, will close this.

@TonyFaith,

you can try below command:
sudo apt-get install --reinstall linux-image-4.4.0-127-generic

closed 2018-06-29 by RealSense-Customer-Engineering