realsenseai/librealsense #1915: install python wrapper failed at make -j4

bhomass, 2018-06-20:
I got thru the instructions for installing python wrapper all the way to make -j4. It says 

make: *** No targets specified and no makefile found.  Stop.

ev-mp, 2018-06-20:
Hi @bhomass , if you start in librealsense root and follow the [instructions](https://github.com/IntelRealSense/librealsense/tree/development/wrappers/python#building-from-source), then  when reaching 'make -j' stage would put you:

--| .        .        .     .     .     . (librealsense root dir)
....| .github
....| _build_
     .     .     .     .   .   .|.   **here**     
...
....| include
....| src
....| scripts
 ...

bhomass, 2018-06-20:
you mean the build subdirectory? I am in that exact location, and make -j4 gives me that error about no makefile. Should I have a makefile there by this stage? There are some .cmake files though. But make command is not using them.

TheMikeyR, 2018-06-21:
@bhomass did you do following before make in the build folder `cmake ../ -DBUILD_PYTHON_BINDINGS=bool:true` ?

bhomass, 2018-06-21:
yes, I did. It completes with no error. The response is as follows

-- Info: REALSENSE_VERSION_STRING=2.12.0
CMake Warning at CMakeLists.txt:576 (message):
  Using libuvc!

-- pybind11 v2.2.1
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/bruceho/workspace/librealsense/build

But still no makefile created. The directory listing is

CMakeCache.txt			cmake_uninstall.cmake		log.txt				tools
CMakeFiles			config				realsense2Config.cmake		wrappers
CMakeScripts			examples			realsense2ConfigVersion.cmake
cmake_install.cmake		librealsense2.xcodeproj		third-party

RealSense-Customer-Engineering, 2018-06-25:
[Realsense Customer Engineering Team Comment]
hi @bhomass,

would like to check some basic for this topic,
what's your OS? Ubuntu? version?
thanks.

bhomass, 2018-06-26:
@TonyFaith no, I just mean you need to create a different build directory for the wrapper code (such as python_build), which is distinct from the build directory you may have used to build the repo at the root level. If you didn't try building at root level, then nothing to worry about. 

@customer-engineering I have OSX, but my problem and solution is unrelated to which OS.

closed 2018-07-23 by RealSense-Customer-Engineering