realsenseai/librealsense #1919: problems in rs_convert

marcovs, 2018-06-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.09.09.02 | 
| Operating System & Version |   MacOS  | 
| Kernel Version (Linux Only)    |                                | 
| Platform                                 | Mac  |
| SDK Version                            |  2.11.0                          | 
| Language                            |  {C++ }                          | 

I tested the recently released version of the rs_convert tool. However I found the following problems:

1. Even with small bag files (~150MB), it drops frames. I was unable to extract all frames from a bag with 427 color and 430 depth frames. The number of frames dropped varies in each execution instance.

2. Sometimes it crashes before completion. I suspect it is also because of timing issues. When debugging, it points to the following line:

```
void Writer::initializeLogger(const std::string& loggerId, bool lookup, bool needLock) {
  if (lookup) {
    m_logger = ELPP->registeredLoggers()->get(loggerId, ELPP->hasFlag(LoggingFlag::CreateLoggerAutomatically));
  }
```

3. It does not seem to have a smooth execution. Sometimes it gets stuck at a certain completion percentage and then it gets a burst of speed.

I was hoping to use the tool as a base for non-realtime processing of rosbags frame-by-frame. However, every time I tried with playback.resume() and playback.pause() even before the rs_convert tool was released I ran in issues 1 or 2. Are there any suggestions for my use-case?

baptiste-mnh, 2018-06-22:
Hi!
I got the exact same issue.

Julz-s, 2018-06-22:
I have this frame dropping issue too.

I recorded about 30 seconds @ 30 fps but instead of the expected 900 depth images I got only 390-400 (exact number varies each time I execute the tool)

TheMikeyR, 2018-06-25:
If you need something temporarily, you can use this python script, it requires opencv and realsense python bindings. You could probably get past opencv by using another image library like Pillow (PIL). 

```
import argparse

import pyrealsense2 as rs
import numpy as np
import cv2
import os

def main():
    if not os.path.exists(args.directory):
        os.mkdir(args.directory)
    try:
        config = rs.config()
        if LOAD_BAG:
            rs.config.enable_device_from_file(config, args.input, False)
        pipeline = rs.pipeline()
        config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)
        pipeline.start(config)
        i = 0
        while True:
            print("Saving frame:", i)
            frames = pipeline.wait_for_frames()
            depth_frame = frames.get_depth_frame()
            depth_image = np.asanyarray(depth_frame.get_data())
            cv2.imwrite(args.directory + "/" + str(i).zfill(6) + ".png", depth_image)
            i += 1
    except RuntimeError:
        print("No more frames arrived, reached end of BAG file!")

    finally:
        pass

if __name__ == "__main__":
    parser = argparse.ArgumentParser()
    parser.add_argument("-d", "--directory", type=str, help="Path to save the images")
    parser.add_argument("-i", "--input", type=str, help="Bag file to read")
    args = parser.parse_args()

    main()
```

baptiste-mnh, 2018-06-25:
Thanks, it looks stable way more stable with the python wrapper!
I keep the issue open as long as the rs-convert still have this bug.

baptiste-mnh, 2018-06-25:
## Objetive
Get the depth image and the color image in two separate folder.

## Code
```
import argparse

import pyrealsense2 as rs
import numpy as np
import cv2
import os

LOAD_BAG = True
SAVE_DEPTH = True
SAVE_RGB = True

def main():
    if not os.path.exists(args.directory):
        os.mkdir(args.directory)
    try:
        config = rs.config()
        if LOAD_BAG:
            rs.config.enable_device_from_file(config, args.input, False)
        pipeline = rs.pipeline()
        if SAVE_DEPTH:
            config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)
        if SAVE_RGB:
            config.enable_stream(rs.stream.color, 1280, 720, rs.format.rgb8, 30)
        pipeline.start(config)
        i = 0
        while True:
            print("Saving frame:", i)
            frames = pipeline.wait_for_frames()
            if SAVE_DEPTH:
                depth_frame =  frames.get_depth_frame()
                depth_image = np.asanyarray(depth_frame.get_data())
                depth_image[np.where((depth_image > 3000))] = 0
                cv2.imwrite(args.directory + "/depth/" + str(i).zfill(6) + ".png", depth_image)
            if SAVE_RGB:
                color_frame = frames.get_color_frame()
                color_image = np.asanyarray(color_frame.get_data())
                cv2.imwrite(args.directory + "/images/" + str(i).zfill(6) + ".png", color_image)
            i += 1
    except RuntimeError:
        print("No more frames arrived, reached end of BAG file!")

    finally:
        pass

if __name__ == "__main__":
    parser = argparse.ArgumentParser()
    parser.add_argument("-d", "--directory", type=str, help="Path to save the images")
    parser.add_argument("-i", "--input", type=str, help="Bag file to read")
    args = parser.parse_args()

    main()
r
```
## Issues with the python exemple
When I only save depth OR color frame, the number of exported images vary but i got ~1400 frames for my ros_bag.
So I lunch it twice (one for depth, one for color frame) and I obtain ~ 2800 frames.

When I try to save both frames at the same time, it also vary but I got less frames (~700 frames with the same ros_bag). So ~1400 frames for both captors...

So even with the python wrapper, the issue persist.

marcovs, 2018-06-25:
I agree, it looks like an issue of timing. I have my own c++ code that can save depth and rgb images in two separate instances and frames are not dropped. However, if I try to save them at the same time or if I introduce any delay in the loop (e.g. if I want to do some processing), frames are dropped. 

It looks like pausing or resuming the playback does not always work, and the realtime flag has no effect

baptiste-mnh, 2018-06-25:
Does your code save the images from a rosbag?
Could you share it please? :)

marcovs, 2018-06-25:
There you go, it's ugly and uses a fat global variable but it works. The function to call is extractImagesFromBagFile. You will need opencv. Let me know how you get on. It also outputs timestamps, feel free to remove it if you don't need it

```
#include "realsenseUtils.hpp"

using namespace rs2;

//global variable to check the status of transcribed video - have to modify
int realsenseProcessedFrames = 0;

void deviceInfo(const device &device) {
    //print playback device configuration
    VLOG(1) << "Device information: ";
    for (int i = 0; i < static_cast<int>(RS2_CAMERA_INFO_COUNT); i++)
    {
        rs2_camera_info info_type = static_cast<rs2_camera_info>(i);
        //SDK enum types can be streamed to get a string that represents them
        VLOG(1) << "  " << std::left << std::setw(20) << info_type;

        //A device might not support all types of RS2_CAMERA_INFO.
        //To prevent throwing exceptions from the "get_info" method we first check if the device supports this type of info
        if (device.supports(info_type))
            VLOG(1) << device.get_info(info_type);
        else
            VLOG(1) << "N/A";
    }

    //get sensors
    std::vector<sensor> sensors = device.query_sensors();
    VLOG(1) << "Device consists of " << sensors.size() << " sensors:\n";
    int index = 0;
    // We can now iterate the sensors and print their names
    for (sensor sensor : sensors)
    {
        std::string sensorName;
        if (sensor.supports(RS2_CAMERA_INFO_NAME))
            sensorName = sensor.get_info(RS2_CAMERA_INFO_NAME);
        else
            sensorName = "Unknown Sensor";

        VLOG(1) << "  " << index++ << " : " << sensorName;
    }

}

void outputStreamInfo(const sensor &s) {
    // A Sensor is an object that is capable of streaming one or more types of data.
    // For example:
    //    * A stereo sensor with Left and Right Infrared streams that
    //        creates a stream of depth images
    //    * A motion sensor with an Accelerometer and Gyroscope that
    //        provides a stream of motion information

    // Using the sensor we can get all of its streaming profiles
    std::vector<rs2::stream_profile> stream_profiles = s.get_stream_profiles();

    // Usually a sensor provides one or more streams which are identifiable by their stream_type and stream_index
    // Each of these streams can have several profiles (e.g FHD/HHD/VGA/QVGA resolution, or 90/60/30 fps, etc..)
    //The following code shows how to go over a sensor's stream profiles, and group the profiles by streams.
    std::map<std::pair<rs2_stream, int>, int> unique_streams;
    for (auto&& sp : stream_profiles)
    {
        unique_streams[std::make_pair(sp.stream_type(), sp.stream_index())]++;
    }
    VLOG(1) << "Sensor consists of " << unique_streams.size() << " streams: ";
    for (size_t i = 0; i < unique_streams.size(); i++)
    {
        auto it = unique_streams.begin();
        std::advance(it, i);
        VLOG(1) << "  - " << it->first.first << " #" << it->first.second;
[... 214 lines cut ...]
            break;
    }
    depthSensor.stop();
    depthSensor.close();
    fout.close();
    
    //process RGB frames
    realsenseProcessedFrames = 0;
    fout.open(string(outFolder).append("/rgbTimestamp.txt"), std::ios_base::out);
    rgbSensor.open(rgbStreamProfile);
    rgbSensor.start([&](frame f) {displayRGBFrame(f, fout, outFolder);});
    while (true) {
        int prevProcessedFrames = realsenseProcessedFrames;
        std::this_thread::sleep_for(std::chrono::milliseconds(1000));
        if (prevProcessedFrames == realsenseProcessedFrames)
            break;
    }
    rgbSensor.stop();
    rgbSensor.close();
    fout.close();
    
    realsenseProcessedFrames = 0;
}

```

baptiste-mnh, 2018-06-25:
I'll test this as soon as I can and come back to you! Thanks!

baptiste-mnh, 2018-07-10:
Hi @marcovs, I've an issue with your code, I only have few .png in the output folder but  the realsenseProcessedFrames indicate the good number of frames.

I think my declaration of the fout variable is wrong (I think yours is in your realsenseUtils.hpp).

Here is what I did in the extractImagesFromBagFile function : 
```
void extractImagesFromBagFile(const string &filename, const string &outFolder) {
    // Declare depth colorizer for pretty visualization of depth data
    colorizer color_map;
    color_map.set_option(RS2_OPTION_COLOR_SCHEME, 1);
    color_map.set_option(RS2_OPTION_HISTOGRAM_EQUALIZATION_ENABLED,true);
    rs2::align align_to(RS2_STREAM_COLOR);
    
    //Declare bag file in device configuration
    context ctx;
    playback dev = ctx.load_device(filename);
    dev.set_real_time(false);
    deviceInfo(dev);
    
    //set sensors
    vector<sensor> sensors = dev.query_sensors();
    //std::cout << "Select RGB Module Index" << std::endl;
    //uint32_t usrInput;
    //std::cin >> usrInput;
    sensor rgbSensor = sensors[1];
    std::cout << "RGB sensor: " << rgbSensor.get_info(RS2_CAMERA_INFO_NAME) << std::endl;
    outputStreamInfo(rgbSensor);
    stream_profile rgbStreamProfile = rgbSensor.get_stream_profiles()[0];
    cout << "Video stream: " << rgbStreamProfile.stream_type() << ", " << rgbStreamProfile.stream_name() << " " << rgbStreamProfile.format() << " " << rgbStreamProfile.as<video_stream_profile>().width() << "x" << rgbStreamProfile.as<video_stream_profile>().height() << "@" << rgbStreamProfile.fps() <<endl;
    
    //std::cout << "Select Depth Module Index" << std::endl;
    //std::cin >> usrInput;
    sensor depthSensor = sensors[0];
    std::cout << "Depth stream: " << depthSensor.get_info(RS2_CAMERA_INFO_NAME) << std::endl;
    outputStreamInfo(depthSensor);
    stream_profile depthStreamProfile = depthSensor.get_stream_profiles()[0];
    cout << "Depth stream: " << depthStreamProfile.stream_type() << ", " << depthStreamProfile.stream_name() << " " << depthStreamProfile.format() << " " << depthStreamProfile.as<video_stream_profile>().width() << "x" << depthStreamProfile.as<video_stream_profile>().height() << "@" << depthStreamProfile.fps() << endl;
    
    //open profiles and start streaming

   /////////////////////////////////////////////////////////////
   ofstream fout; // FOUT DECLARATION
  /////////////////////////////////////////////////////////////

    ofstream.open(string(outFolder).append("/depthTimestamp.txt"), std::ios_base::out);
    depthSensor.open(depthStreamProfile);
    depthSensor.start([&](frame f) {displayDepthFrame(f, fout, color_map, outFolder);});
    while (true) {
        int prevProcessedFrames = realsenseProcessedFrames;
        std::this_thread::sleep_for(std::chrono::milliseconds(1000));
        if (prevProcessedFrames == realsenseProcessedFrames)
            break;
    }
    depthSensor.stop();
    depthSensor.close();
    fout.close();
    
    //process RGB frames
    realsenseProcessedFrames = 0;
    fout.open(string(outFolder).append("/rgbTimestamp.txt"), std::ios_base::out);
    rgbSensor.open(rgbStreamProfile);
    rgbSensor.start([&](frame f) {displayRGBFrame(f, fout, outFolder);});
    while (true) {
        int prevProcessedFrames = realsenseProcessedFrames;
        std::this_thread::sleep_for(std::chrono::milliseconds(1000));
        if (prevProcessedFrames == realsenseProcessedFrames)
            break;
    }
    rgbSensor.stop();
    rgbSensor.close();
    fout.close();
    
    realsenseProcessedFrames = 0;
}
```

baptiste-mnh, 2018-07-10:
I finally successfully ran your code but I still got frame drop.
After each run, I got a different number of frames output.

Here is what i did to save all the frames (my issues was that the name of the frame saved was the same at each iteration): 
```
void displayRGBFrame(frame f, ofstream &fout, const std::string &baseFolder) {
    cv::Mat rgbMat = frame_to_mat(f);
    
    std::string frameNoString = "Frame Counter";
    std::string timestampString = "Time Of Arrival";
    rs2_metadata_type frameNo = 0;
    rs2_metadata_type timestamp = 0;
    std::string outputFilename;
    for (size_t i = 0; i < RS2_FRAME_METADATA_COUNT; i++)
    {
        if (f.supports_frame_metadata((rs2_frame_metadata_value)i))
        {
            std::string property = rs2_frame_metadata_to_string((rs2_frame_metadata_value)i);
            rs2_metadata_type value = f.get_frame_metadata((rs2_frame_metadata_value)i);
            if (frameNoString.compare(property) == 0) {
                frameNo = value;
                std::stringstream ss;
                ss << value;
                ss >> outputFilename;
            }
            if (timestampString.compare(property) == 0) {
                timestamp = value;
            }
            std::cout << rs2_frame_metadata_to_string((rs2_frame_metadata_value)i) << "," << value << "\n";
        }
    }
    
    //output values and frame
    try {
        string path = std::string(baseFolder).append("/colour/colour_"+to_string(realsenseProcessedFrames)).append(outputFilename).append(".png");
        cout << "------------------------------------------------\n"<< path << endl;
        cv::imwrite(path, rgbMat);
        realsenseProcessedFrames++;
    } catch(cv::Exception &ex) {
        cout << "Error when writing depth image to png: " << ex.what() << endl;
    }
    
    fout << frameNo << " " << timestamp << std::endl;

}

void displayDepthFrame(frame f, ofstream &fout, const colorizer &color_map, const std::string &baseFolder) {
    
    cv::Size depthSize;
    depthSize.width = f.as<video_frame>().get_width();
    depthSize.height = f.as<video_frame>().get_height();
    frame depthColor = color_map(f);
    cv::Mat depthMat(cv::Size(1280, 720), CV_16U, (void*) depthColor.get_data());
    //cv::Mat depthMat(depthSize, CV_16U, (void*)f.get_data());
    
    
    std::string frameNoString = "Frame Counter";
    std::string timestampString = "Time Of Arrival";
    rs2_metadata_type frameNo = 0;
    rs2_metadata_type timestamp = 0;
    std::string outputFilename;
    for (size_t i = 0; i < RS2_FRAME_METADATA_COUNT; i++)
    {
        if (f.supports_frame_metadata((rs2_frame_metadata_value)i))
        {
            std::string property = rs2_frame_metadata_to_string((rs2_frame_metadata_value)i);
            rs2_metadata_type value = f.get_frame_metadata((rs2_frame_metadata_value)i);
            if (frameNoString.compare(property) == 0) {
                frameNo = value;
                std::stringstream ss;
                ss << value;
                ss >> outputFilename;
            }
            if (timestampString.compare(property) == 0) {
                timestamp = value;
            }
            std::cout << rs2_frame_metadata_to_string((rs2_frame_metadata_value)i) << "," << value << "\n";
        }
    }
    
    //output values and frame
    try {
        string path = std::string(baseFolder).append("/depth/depth_"+to_string(realsenseProcessedFrames)).append(outputFilename).append(".png");
        cout << "------------------------------------------------\n"<< path << endl;

        imwrite( path, depthMat);
        fout << frameNo << " " << timestamp << std::endl;
        realsenseProcessedFrames++;
    } catch(cv::Exception &ex) {
        cout << "Error when writing depth image to png: " << ex.what() << endl;
    }

}
```

apoorva2398, 2018-07-13:
See https://communities.intel.com/message/553535#553535.

marcovs, 2018-08-08:
@baptiste-mnh @apoorva2398 Thank you for the pointers. I worked on it a little bit and I made a class that reads bag files using directly the rosbag API. It implements the following functionalities

1. Reads a bag file and outputs a pair of rgb and depth frames asynchronously whenever the function nextFrame is called. The class does not use the realsense pipeline, rather it parses the rosbag messages, so there are no frames dropped

2. Since AFAIK there is no hardware synchronization between rgb and depth streams on the device, the class checks the timestamps and outputs the pair of rgb and depth files that are closest in time (without expensive searches).

3. The depth frame returned is aligned to the camera point of view

4. Depth frame is saved as an xml with the full floating point values in meters for each pixel

4. Together with the frames, the class outputs the frame number and the timestamp for each frame recorded in the rosbag (according to the system clock)

The code should be quite readable, and the dependencies are opencv 3.x, glog, librealsense, and the part of the ros library included as headers, and boost. The files print_helpers.cpp/h are a modified version of the ones included in the rosbag viewer tool example. 

Even if you don't use the class directly, you should be able to copy and paste the methods inside the class to read the rosbag however you want.

[RealSenseHelper.zip](https://github.com/IntelRealSense/librealsense/files/2270032/RealSenseHelper.zip)

Hope it helps, let me know how you get on.

@ev-mp I hope it can be useful

dorodnic, 2018-09-25:
@baptiste-mnh @TheMikeyR @marcovs - I think ROS-bag reading issues are behind us, with [v2.16.1](https://github.com/IntelRealSense/librealsense/releases/tag/v2.16.1) release.

RealSense-Customer-Engineering, 2018-10-05:
[Realsense Customer Engineering Team Comment]
Hi @marcovs, @baptiste-mnh,

Any update based on the new librealsense?

closed 2018-11-03 by dorodnic

YangJiao1996, 2019-01-26:
Hi @dorodnic @RealSense-Customer-Engineering,
I believe the frame dropping and crashing issues still exist even with the latest librealsense-2.18.0.