realsenseai/librealsense #1920: Segmentation fault with OpenCV when enabling color stream

javierselva, 2018-06-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |   Linux (Ubuntu 16)   | 
| Kernel Version (Linux Only)    |  4.13.0-43-generic                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.10.3                     | 
| Language                            | C++      | 

### Issue Description
I've been struggling with this for a while. I don't know if this is expected behavior or not, so I thought I'd share it.
I was playing with the [imshow](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/opencv/imshow/rs-imshow.cpp) example. I replaced every "depth" in the code to "color" and `rs2::frame` to `rs2::video_frame`. Then I decided to use a configuration and just added `cfg.enable_stream(RS2_STREAM_COLOR)` 
The resulting code is as follows:

>     #include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
>     #include <opencv2/opencv.hpp>   // Include OpenCV API
>     int main(int argc, char * argv[])
>     {
> 
>         rs2::config cfg;
> 
>         //Add desired streams to configuration
>         cfg.enable_stream(RS2_STREAM_COLOR);
> 
>         rs2::pipeline pipe;
> 
>         // Start streaming with default recommended configuration
>         pipe.start(cfg);
> 
>         using namespace cv;
>         const auto window_name = "Display Image";
> 
>         while (true)
>         {
>             rs2::frameset data = pipe.wait_for_frames(); // Wait for next set of frames from the camera
>             rs2::video_frame color = data.get_color_frame();
> 
>             // Query frame size (width and height)
>             const int w = color.get_width();
>             const int h = color.get_height();
> 
>             // Create OpenCV matrix of size (w,h) from the colorized color data
>             Mat image(Size(w, h), CV_8UC3, (void*)color.get_data(), Mat::AUTO_STEP);
>             // Update the window with new data
>             imshow(window_name, image);
> 
>             char c = (char) cv::waitKey(25);
>             if(c==27) break;
>         }
> 
>         return EXIT_SUCCESS;
>     }
> 

This code as is rises a _Segmentation fault (core dumped)_ when reaching the `imshow` line. This does not happen with the original example when adding RS2_STREAM_DEPTH nor if no configuration is used. In order to fix this I had to add a second parameter: `cfg.enable_stream(RS2_STREAM_COLOR,RS2_FORMAT_BGR8);`.

My guess is that when only asking for the Color Stream the remaining default parameters are such that imshow is unable to handle the resulting frame, but I'm not quite sure why. Also, as I said above, I don't really know if this is actually expected behavior or not, but it got me nuts for a while.

Hope this helps somebody!

RealSense-Customer-Engineering, 2018-06-26:
[Realsense Customer Engineering Team Comment]
@javierselva
since you only enable stream with type specified but not the format, so it will return the default value for you which is "RS2_FORMAT_RAW16" that's the reason why you encounter the problem. To fix the problem is simple, just specify which format you need, so change your line of "cfg.enable_stream(RS2_STREAM_COLOR)" to "cfg.enable_stream(RS2_STREAM_COLOR, RS2_FORMAT_BGR)" then everything works well. You can also get back the activated profile to check which format is being configured, I attache the code snippet below for your reference.

`​ rs2::config cfg;`
` //cfg.enable_stream(RS2_STREAM_COLOR); //this will cause seg fault in Mat(...)`
` cfg.enable_stream(RS2_STREAM_COLOR, RS2_FORMAT_BGR8);`
` rs2::pipeline_profile pp = pipe.start(cfg);`
` rs2::stream_profile sp = pp.get_stream(RS2_STREAM_COLOR);`
``

javierselva, 2018-06-26:
I see! So it is just that the default format is not a valid frame for OpenCV.

Thanks a lot, the solution works!

closed 2018-06-26 by javierselva