realsenseai/librealsense #1924: Meaning of published depth values in ROS

msieb1, 2018-06-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D436 | 
| Operating System & Version |   Ubuntu 14 | 
| Platform                                 | PC |
| Language                            |  ROS                       | 

### Issue Description
I am using the realsense2_camera launch files and I am publishing the /camera/aligned_depth_to_color/image_raw channel.
Teh depth values are given in Z16 format from [0,64). I am wondering what the actual resolution is since I cannot find that anywhere. When using the pyrealsense2 library in Python, I can query the depth scale to convert the Z16 values to meaningful metric values. Is it safe to assume the same scale is used for the values published in ROS?

RealSense-Customer-Engineering, 2018-06-22:
[Realsense Customer Engineering Team Comment]
Hi @msieb1,

You can use below command to reinstall kernel:
$ sudo apt-get install --reinstall-Linux-image-4.4.0-127-generic

RealSense-Customer-Engineering, 2018-07-04:
[Realsense Customer Engineering Team Comment]
Hi @msieb1,

You can review the launch file (ex: rs_camera.launch) to see the resolution configured for depth/IR/color.

You can also run below command to configure the camera control values.

$ rosrun rqt_reconfigure rqt_reconfigure

RealSense-Customer-Engineering, 2018-07-16:
[Realsense Customer Engineering Team Comment]
Hi @msieb1,

Still need any support for this topic?

RealSense-Customer-Engineering, 2018-07-23:
[Realsense Customer Engineering Team Comment]
Hi @msieb1,

I will close this issue first. If any more question, you still can re-open the issue or create another ticket.

closed 2018-07-23 by RealSense-Customer-Engineering