realsenseai/librealsense #1930: Cannot activate D415

Arnandsway, 2018-06-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D400  Firmware 5.9.11} | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win 10) / | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<12>.<0> }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I cannot activate the D415 when starting the RealSense Viewer. Not after rebooting etc.
The weird thing is, it did work before an Windows Update. 
Since the latest release, 1803, it doesn't work.

This is the error:

Unknown in rs2_create_device(info_list:07D1DC88, index:0):
Could not initialize extensions controls!

nachoreimat, 2018-06-21:
Hey same problem happened to me.
It is because of the last windows update 1803
It turns off the privacy settings for accessing the camera and microphone.
You can go to Windows<Settings<Privacy<Camera and change the toggle to allow apps using the camera.

RealSense-Customer-Engineering, 2018-06-25:
[Realsense Customer Engineering Team Comment]
according to the error message provided
"Unknown in rs2_create_device(info_list:07D1DC88, index:0):
 Could not initialize extensions controls! "

The issue should due to the setting for app to access the camera is turn off as nachoreimat also commented
the in box windows camera app will also ask your permission while launch it
please see if turn it on can fix your problem.

RealSense-Customer-Engineering, 2018-07-02:
[Realsense Customer Engineering Team Comment]
Hi @Arnandsway
do you still have issue after grant the app to access the camera?
best

closed 2018-08-31 by RealSense-Customer-Engineering