realsenseai/librealsense #1933: SR300 depth stream doesn't work

Kebsox, 2018-06-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  SR300 } | 
| Firmware Version                   | 3.21.0.0 | 
| Operating System & Version |    Linux (Ubuntu 16.04 | 
| Kernel Version (Linux Only)    |  4.13.0-45-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  1.12}                          | 
| Language                            |  {C++ }                          | 

### Issue Description

Hello everybody,

I have a problem when using my SR300 camera. My depth data stream is empty. When I try to display it only a black window appears. When I want to record the value of the feed I only have zeros. THE code works very well with my R200 camera. Where it starts to get strange is that the SR300 works very well with RVIZ. Has anyone ever had a similar problem? 

Thanks you for reading this.
Excuse my english is not my first language

RealSense-Customer-Engineering, 2018-06-22:
[Realsense Customer Engineering Team Comment]
Hi @Kebsox,

Would you mind to use SDK 2.x instead of SDK 1.12? SDK 2.x also support SR300 stream.

Kebsox, 2018-06-22:
Hi,

Yes i can try with the 2.x , but can i use it on ROS?

RealSense-Customer-Engineering, 2018-06-25:
[Realsense Customer Engineering Team Comment]
Hi @Kebsox,

Yes, you can use ROS + librealsense 2.x. Find more info in below link:
https://github.com/intel-ros/realsense
The setup configurations can be ros wrapper 2.0.3 + ros kinetic + librealsense 2.10.2.

Kebsox, 2018-06-25:
Hi, 

I try your config but i got a mistake i don't know where. I got this message 

[ERROR] [1529924476.821196283]: Failed to load nodelet [/camera/realsense2_camera] of type [realsense2_camera/RealSenseNodeFactory] even after refreshing the cache: API version mismatch: librealsense.so was compiled with API version 2.12.0 but the application was compiled with 2.11.1! Make sure correct version of the library is installed (make install)

I understand the mistake but i don't know how to fix it do you have a clue?

Thanks you

RealSense-Customer-Engineering, 2018-06-25:
[Realsense Customer Engineering Team Comment]
Hi @Kebsox,

I think you previously installed librealsense 2.11.1 binaries in /usr/bin. Please uninstall the binaries from /usr/bin. (Go to your librealsense 2.11.1 build folder, and run "sudo make uninstall")
Then, install librealsense 2.12.0 in /usr/bin and re-compile librealsense ROS binaries. Then, you should be able to run those launch files.

Kebsox, 2018-06-25:
Hi,

I just make it works, i will edit my post but you are too fast sorry. I got the corect setup now i will make the change to my code and i will send the result.

Kebsox, 2018-06-25:
Hi again, 

 The realsense2 works i can make this command works roslaunch realsense2_camera rs_camera.launch
But when i want to compile i have a new errors with undefined reference. With librealsense i had librealsense in my catkin, but i can't find what i need to add for realsense2. Just librealsense2 doesn't work sadly.

RealSense-Customer-Engineering, 2018-06-26:
[Realsense Customer Engineering Team Comment]
Hi @Kebsox,

Can you clarify what new errors you see?

Kebsox, 2018-06-26:
Hi, 
i got lot's of errors like this one : Test_rs2.cpp:(.text._ZNK3rs27context13query_devicesEv[_ZNK3rs27context13query_devicesEv]+0x907) : référence indéfinie vers « rs2_get_librealsense_exception_type »

I think a have to add a dependency in my catkin.

RealSense-Customer-Engineering, 2018-06-27:
[Realsense Customer Engineering Team Comment]
Hi @Kebsox,

As you mentioned, below command works. You may need to further check your development environment to resolve those errors you saw.
$ roslaunch realsense2_camera rs_camera.launch

If no other question about this issue topic, will close this. You still can re-open or create another ticket if you have any further question.

closed 2018-06-27 by Kebsox