realsenseai/librealsense #1934: Auto white-balance and exposure

yanglin5689446, 2018-06-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Ubuntu 16.04}  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {++ }                          | 

### Issue Description

Hello, 
I have a question about auto white-balance and exposure.
In a legacy version of librealsense (which is used in [ElasticFusion](https://github.com/mp3guy/ElasticFusion/blob/master/GUI/src/Tools/RealSenseInterface.cpp) ) it seems that the white-balance and exposure need to be set to auto with set_option API,
But in current version, this API seems to be removed and the comment [here]
(https://github.com/IntelRealSense/librealsense/blob/efe67a9e7b69c7a8c3657456543b1a220095d08b/include/librealsense2/h/rs_option.h#L27) says that in current version if we don't modify the option, the auto white-balance and exposure are enabled by default, are they?

Thank you for any help you can provide!

ev-mp, 2018-06-25:
Both auto-white balance and auto-exposure controls are supported in librealsense SDK 2.0 via the same API that was available in SDK v1.0 : `rs2_get_option(...)` and `rs2_set_option(...)`.

Note that the white-balance is RGB-only control, and applies exclusively to D415, D435 and SR300.

The default values of those controls are preset in the camera's firmware.
You can run `rs-enumerate-devices` utility to read those default for the cameras you have.
For instance, D435 with FW 5.9.11 ( [currently recommended](https://downloadcenter.intel.com/download/27792/Latest-Firmware-for-Intel-RealSense-D400-Product-Family?v=t)):

 
Options for **Stereo Module**
 Supported options:                                    min        max       step  **default**
    Exposure                                           : 20   ... 166000      20    8500
    **Enable Auto Exposure**                               : 0    ... 1           1     **1**

Options for **RGB Camera**
 Supported options:                                    min        max       step  **default**
    Exposure                                           : 39   ... 10000       39    156
    White Balance                                      : 2800 ... 6500        10    4600
    **Enable Auto Exposure**                               : 0    ... 1           1     **1**
    **Enable Auto White Balance**                          : 0    ... 1           1     **1**

RealSense-Customer-Engineering, 2018-07-02:
[Realsense Customer Engineering Team Comment]
Hi @yanglin5689446
as ev-mp's comment that you can use the sample app rs-enumerate-devices to list the default setting
and same API still valid in case you need to change those setting
please let us know if this can help on your issue

RealSense-Customer-Engineering, 2018-07-23:
[Realsense Customer Engineering Team Comment]
​Dear @yanglin5689446
any further question related to this report issue?

closed 2018-07-30 by RealSense-Customer-Engineering