realsenseai/librealsense #1935: Map one Depth Point to Color Frame

DvdZmd, 2018-06-21:
| Required Info                          |                                         |
|---------------------------------|--------------------------- |
| Camera Model                        | D415/D435                     | 
| Firmware Version                    | 05.09.02.00                     | 
| Operating System & Version  |   Win 10                          |  
| Platform                                  | PC                                   |
| SDK Version                            |  RealSense SDK 2.0         | 
| Language                                |  C                                    | 

Hello, I need to map an specific depth point to the RGB frame. I was reading the rs-exmaples-pointcloud and there it map the entire depth frame to the RGB, can anyone help me? Looking around the doc I don't find what I need, my apologies for start a new thread if the answer was there and I don't see it.

Thanks!

dorodnic, 2018-06-25:
Hi @DvdZmd 
Please take a look in [rsutil.h]()
I think you would need to do something like this:
```cpp
rs2::pipeline pipe;
pipe.start();

auto fs = pipe.wait_for_frames();

auto depth = fs.get_depth_frame();
auto depth_profile = depth.get_profile().as<rs2::video_stream_profile>();
auto depth_intrin = depth_profile.get_intrinsics();

auto color = fs.get_color_frame();
auto color_profile = color.get_profile().as<rs2::video_stream_profile>();
auto color_intrin = color_profile.get_intrinsics();

auto extrin = depth_profile.get_extrinsics_to(color_profile);
// Maybe color_profile.get_extrinsics_to(depth_profile); ??

float xyz[3];
float color_xyz[3];
float ij[2] = { 100, 100 }; // Pixel indexes
float color_ij[2]; // Matching color pixel

rs2_deproject_pixel_to_point(xyz, &depth_intrin, ij, depth.get_distance(ij[0], ij[1]));
rs2_transform_point_to_point(color_xyz, extrin, xyz);
rs2_project_point_to_pixel(color_ij, color_intrin, color_xyz);
```

kargarisaac, 2018-07-02:
Thnaks @dorodnic .
I use python. I recorded some .bag files and when I want to get frames (rgb and depth) I understand that position and even state of objects in rgb and depth frames are different. I want to map an object from rgb to depth. but I think there is a delay between frames or something like this that cause difference between frames.
I also can not get all frames from .bag file. I saw in other Issues that you said wait_for_frames don't get all frames. Is this problem True for online streaming or just for recorded files?
Thanks in advance

RealSense-Customer-Engineering, 2018-07-03:
[Realsense Customer Engineering Team Comment]
@kargarisaac
If you want similar code in Python, there is another issue having the code inside, but do remember to have the last parameter "depth value" be multiply the depth scale unit OR use the get_distance method in depth frame.

Please refer to issue here #1890

kargarisaac, 2018-07-04:
@RealSense-Customer-Engineering 
Thanks. I aligned but the frames are not the same. I think maybe missed frames cause such a problem. I applyobject detection techniques using deep learning on rgb image and find bounding boxes around my object but when I use obtained coordinates of box on depth image, I see a lot of difference. My object moves and It's like the object is 5 to 10 frames back in depth frame rather than rgb frame.
What is the problem you think?

RealSense-Customer-Engineering, 2018-07-12:
[Realsense Customer Engineering Team Comment]
@kargarisaac
Do you still have question on this topic?

kargarisaac, 2018-07-12:
@RealSense-Customer-Engineering 

I updated the FW but still have the problem. when I save the file using realsense viewer it misses some frames and when I use to stream it online the missing rate is higher. about 15 to 20 frames and RGB and depth frames are not synced.

RealSense-Customer-Engineering, 2018-07-16:
[Realsense Customer Engineering Team Comment]
@kargarisaac
How do you save the frames using realsense viewer? do you use recording?
Also how do you calculate the frame drop rate? Can you give more detail description or sample code.

kargarisaac, 2018-07-16:
@RealSense-Customer-Engineering 
Yes, I use the recorder in realsense viewer.

I use your codes exist in GitHub repo and use frame_number method to see the frame number and the numbers have different distances from each other. some numbers such as 5-10-15. for online streaming using python code in the repo, it is higher than the recorded .bag file.

RealSense-Customer-Engineering, 2018-07-16:
[Realsense Customer Engineering Team Comment]
@kargarisaac
Please attach your code revised from sample code in librealsense so I can base on the same baseline to reduplicate your issue. Thank you

closed 2018-07-30 by RealSense-Customer-Engineering