realsenseai/librealsense #1936: CMake compile librealsense with advanced mode APIs

mutp, 2018-06-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.09.09.02 | 
| Operating System & Version |   Win10  | 
| Platform                                 | PC |
| SDK Version                            |  2.10.4                        | 
| Language                            |  C++                         | 

### Issue Description
I tried to compile from source using CMake with Advanced mode APIs using the following command  
`cmake -G "Visual Studio 15 2017 Win64" .. -DBUILD_EXAMPLES=true -DBUILD_RS400_EXTRAS=true`
but I got this output 

> -- The C compiler identification is MSVC 19.13.26132.0
-- The CXX compiler identification is MSVC 19.13.26132.0
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/bin/Hostx86/x64/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/bin/Hostx86/x64/cl.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/bin/Hostx86/x64/cl.exe
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/bin/Hostx86/x64/cl.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Info: REALSENSE_VERSION_STRING=2.10.4
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Failed
-- Found OpenGL: opengl32
-- Configuring done
-- Generating done
-- CMake Warning:
--   Manually-specified variables were not used by the project:
--     BUILD_RS400_EXTRAS
--
-- Build files have been written to: D:/Workspace/Intel Realsense/librealsense-2.10.4/build

It clearly states that `BUILD_RS400_EXTRAS` was not utilized. I also couldn't find **rs400-advanced-mode-sample** example in my build folder.

How do I compile advanced mode APIs for realsense?

ev-mp, 2018-06-21:
Hello @mutp , 
Advanced Mode functionality has been long merged into librealsense core, including v2.10.4.
No additional cmake flags are needed to enable it.
You may disregard that para from the installation guide and build librealsense with the regular CMake flags.

Note that version 2.10.4 is rather outdated. Sticking with it may prevent you from updating the device firmware, as well as taking advantage of multiple improvements and bug fixes that were introduced into librealsense in the last six months.

mutp, 2018-06-22:
Sure, thanks for the info @ev-mp