realsenseai/librealsense #1937: RealSense Repository Key Unavailable, apt-get update fails again

AnGPU, 2018-06-22:
| Camera Model                       | { ZR300 } | 
| Firmware Version                   | (2.0.71.28) | 
| Operating System & Version |   { Linux (Ubuntu 16/)  | 
| Kernel Version (Linux Only)    |  (4.4.0-124-generic)                                         | 
| Platform                                 | PC |
| SDK Version                            |  {Intel® RealSense™ Cross Platform API }                          | 
| Language                            |  {C/}                          | 

There was a problem of connection with the Realsense Repository that was solved by Intel changing the key in #1112 but as it is happening again and that issue is closed I find necessary to reopen it.

ev-mp, 2018-06-25:
Hello @AnGPU , please provide the error log for review.
Thanks

AnGPU, 2018-06-25:
E: Fail to obtain http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/xenial/Release  Input «main/source/Sources» couldn't be found in file «Release» (wrong input in «sources.list» or wrong built file)
E: Some index files couldn't be downloaded, they are skipped or used old files instead.
"translated from spanish log of sudo apt-get update" 
Is this the same problem isn't it?

ev-mp, 2018-06-27:
Hello @AnGPU , the debian repo server configuration has been re-evaluated.
Can you retry and update ?
Thanks

AnGPU, 2018-06-28:
correct from uname -r command I obtain 4.4.0-124-generic 

I tried in another ubuntu and it works so I compare the source list between both and guess there are replicated software, maybe from the last time before the key were updated. Now I uncheck those repos and the update is complete. Thanks for your attention.
[image]

closed 2018-06-28 by AnGPU

suho0515, 2019-11-26:
@AnGPU 
this answer is what i was looking for, and it works for me perfectly.
Thanks!