realsenseai/librealsense #1938: Recording to file from 2 D415 in parallel

mutp, 2018-06-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.09.09.02 | 
| Operating System & Version |   Win10  | 
| Platform                                 | PC |
| SDK Version                            |  2.12.0                        | 
| Language                            |  C++                         | 

### Issue Description
I need to record frames from 2 D415 cameras connected to my PC and write the frames to disk. This needs to happen at the same time in parallel to ensure data is captured at the same time. I have run the following code in different threads, using OpenMP/C++11 but I always unpredictable recorded data. 

```
pipe.stop();

cfg.enable_stream(RS2_STREAM_INFRARED, 1280, 720, RS2_FORMAT_Y8, 30);
cfg.enable_stream(RS2_STREAM_DEPTH, 1280, 720, RS2_FORMAT_Z16, 30);
cfg.enable_stream(RS2_STREAM_COLOR, 1280, 720, RS2_FORMAT_RGB8, 30);

// create ros .bag file on disk for recording
cfg.enable_record_to_file(filename);

// start pipeline of device with desired config
pipe.start(cfg);
```

Since threads will often block each other it will lead to unequal recorded data. 
However, the Realsense Viewer is able to record data from multiple cams very well. Can someone help me how this is done in the Viewer?

zivsha, 2018-06-22:
Im not sure I understand, are you trying to record to the same file?

mutp, 2018-06-22:
No. Recording to two different files.

zivsha, 2018-06-22:
The Viewer is recording using the `rs::recorder` instead of using a pipeline. That's probably the main difference, not sure how can your threads block each other, if they use different instances. In fact, why are you using more than a single thread?

mutp, 2018-06-22:
I tried to use `rs2::recorder`, however it only creates a 5kb file but doesn't write any data to the file. This is the code I have. What am I missing here? Thanks for taking the time to reply.
```
std::shared_ptr<rs2::recorder> _recorder;
_recorder = std::make_shared<rs2::recorder>(filename1, view_port1.device);
Sleep(10000);
_recorder.reset();
```

zivsha, 2018-06-22:
Did you call `enable_stream` and `start` on `view_port1.device`?
Regarding the multiple threads part, well you would also get some delay between the two. I don't think you should worry about recording at the same time, but more about how to start the two playbacks at the same delay as you have between starting record of one device and the other.
If you need "perfect" timing, you might want to think of using a recorder on a software_device you create which you will pass all frames from both devices, this way you will get a single file containing frames from both devices, but will lose part of the data per device (like options and info)

mutp, 2018-06-23:
```
pipe1.stop();
cfg1.enable_stream(RS2_STREAM_INFRARED, 1280, 720, RS2_FORMAT_Y8, 30);
cfg1.enable_stream(RS2_STREAM_DEPTH, 1280, 720, RS2_FORMAT_Z16, 30);
cfg1.enable_stream(RS2_STREAM_COLOR, 1280, 720, RS2_FORMAT_RGB8, 30);
pipe1.start(cfg1);
_recorder = std::make_shared<rs2::recorder>(filename1, view_port1.device);
Sleep(10000);
_recorder.reset();
```

This is the code I'm running. It still outputs a 5KB file. 
I hadn't thought of using a software_device. I could try that but any idea why the above code doesn't work?

zivsha, 2018-06-24:
Not sure why it doesn't work, can you try the following first: (since i'm not sure where `view_port1.device` is from):
pipe1.stop();
cfg1.enable_stream(RS2_STREAM_INFRARED, 1280, 720, RS2_FORMAT_Y8, 30);
cfg1.enable_stream(RS2_STREAM_DEPTH, 1280, 720, RS2_FORMAT_Z16, 30);
cfg1.enable_stream(RS2_STREAM_COLOR, 1280, 720, RS2_FORMAT_RGB8, 30);
auto profile1 = pipe1.start(cfg1);
_recorder = std::make_shared<rs2::recorder>(filename1, profile1.get_device());
Sleep(10000);
_recorder.reset();

mutp, 2018-06-26:
Your code worked. Thanks :) 
But, it doesn't work if I store the "rs2::device device_" as a private global variable in the constructor and use this variable in `_recorder = std::make_shared<rs2::recorder>(filename1, device_);`
It seems very strange.

zivsha, 2018-06-26:
Sorry for repeating myself but since you only share bits of you code and logic at a time I can't know for sure what it is you are doing wrong, or doing right and where the bug is.
I don't see an issue with storing a `rs2::device` as a private member. Many of the samples use a this as a "global" (in the sample it's just on the main stack so not really global) variable. 
The main thing you should understand is that `rs2::device` acts more as a shared pointer than a value type, and that the pipeline is free to choose a device according the the demands in the config object passed to it (or any device is not config is passed). The only time you know which device the pipeline is using is between `pipeline.start()` until `pipeline.stop()`, and you can get this device the either by using the returned profile from `start()` or by querying for `pipline.get_active_profile().get_device()` (not sure about the API, but something like it).

mutp, 2018-06-27:
That's very helpful. Thanks for taking the time @zivsha!

mutp, 2018-06-27:
I'm still struggling with this. 
This is the code I'm using - 

File1.cpp
```
int main()
{
---
  rs_device_custom connected_devices;
  rs2::device_list devices = ctx.query_devices();
  connected_devices.record_device(devices[0]);
---
}
```
File2.cpp
```
void rs_device_custom::record_device(rs2::device device)
{
	std::string serial_number(device.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));

	rs2::pipeline pipeline;
	rs2::config cfg;
	std::shared_ptr<rs2::recorder> _recorder;

	std::string dir = "D:/Workspace/Intel Realsense/proj/data/streams/";
	std::string filename;
	std::string date_time = tv_utils::currentDateTime();

	filename = dir + serial_number + "_" + date_time + std::string(".bag");

	cfg.enable_stream(RS2_STREAM_INFRARED, 1280, 720, RS2_FORMAT_Y8, 30);
	cfg.enable_stream(RS2_STREAM_DEPTH, 1280, 720, RS2_FORMAT_Z16, 30);
	cfg.enable_stream(RS2_STREAM_COLOR, 1280, 720, RS2_FORMAT_RGB8, 30);
	cfg.enable_device(serial_number);

	auto profile = pipeline.start(cfg);

	_recorder = std::make_shared<rs2::recorder>(filename, device);
	Sleep(10000);
	_recorder.reset();
}
````
This code does not work. 
However, if I change this line from 
`_recorder = std::make_shared<rs2::recorder>(filename, device);` 
to 
`_recorder = std::make_shared<rs2::recorder>(filename, profile.get_device());` the file is recorded to disk.

If I just use the function parameter `device`, the file recorded is always a 5kb file.
What am I doing wrong here? Sorry if I'm missing something obvious.

mutp, 2018-06-27:
I also have another question. 
You said I can the current device using the API `profile.get_device()`. This requires me to use the pipeline workflow.
If I'm not using a pipeline to stream frames from the device, and instead I use 
`sensor.start([&](rs2::frame f) {  });`, how do I get the current device?

zivsha, 2018-06-28:
Regarding the first question, I'm not sure if this is a bug or expected (yet not obvious) behavior. The pipeline is probably choosing the same underlying device, but I'm guessing that a different instance of a device is used by the pipeline which probably causes some of the device's state to be missing in the device instance you pass to the recorder. I would leave this one for the LibRS team to look at and decide if it should be fixed or if it is even a bug(@dorodnic ).

Anyways, if you are using the pipeline you do not need the `rs2::recorder` and you can use `cfg.enable_record_to_file(filename);`. I only mentioned it since the Viewer is using it but the viewer is not using the pipeline either. If you are still having issue with this one, please share the code that fails and we'll see about that.

If you are not using the pipeline, you are using `rs2::context` which allows you to query `rs2::device` which allows you to query its `rs2::sensor`s. See the [sensor-control](https://github.com/IntelRealSense/librealsense/tree/master/examples/sensor-control) sample on how to use it.

mutp, 2018-07-02:
Okay, sure! Look forward to some clarity on the first question. Thanks for everything else!

RealSense-Customer-Engineering, 2018-07-16:
[Realsense Customer Engineering Team Comment]
@mutp
The main difference is the **actual binding** between user configuration and the underlying hardware that satisfies that.
When the configuration is passed onto pipeline, the pipeline resolves it internally and assign a HW camera to handle it.
Retrieving device from pipeline’s profile will guarantee device validity.

From your code, you should also check if the device sent from external is the same with the one you configured with serial number.
1- Requested the pipeline to mach a specific device with serial number:
…
_cfg.enable_device(serial_number);_
auto profile = pipeline.start(cfg);

2- Then tried  recording rosbag from a device received externally
_recorder = std::make_shared<rs2::recorder>(filename, _device_);

There is no check that those two device are actually the same.

closed 2018-07-23 by RealSense-Customer-Engineering