realsenseai/librealsense #1939: get_texture_coordinates() returns array filled with 0 values

eMrazSVK, 2018-06-22:
| Required Info                         
| Camera Model                        D415
| Firmware Version                    05.09.11.00 
| Operating System & Version   Ubuntu 16 
| Kernel Version (Linux Only)    4.4.38
| Platform                                   Jetson TX2
| SDK Version                            2.0 
| Language                                C++ 
| Segment			          Robotics 

Hello,

I am developing an application, where I would like to use PCL library and so is that, I need to save point cloud data from realsense D415 to .PCD format. This has been discussed here and also in the community quite a lot, but I ran into a strange behavior. 

Here is the link to question I have asked in community forum, the link points to the code which is supposed to work /- [https://communities.intel.com/message/551252#551252](url). I have successfully converted depth frame to PCD format, but only the XYZ values. As far as my application is concerned, I need RGB values too. So I want to add RGB values to each point in the pointcloud. What is actually happening is that each point has same color. I tried a little bit of debugging and what I have found is that point coordinates which are calculated at [https://github.com/eMrazSVK/JetsonSLAM/blob/master/pcl_testing.cpp#L24](url) are always zero. (I printed it like std::cout<<x<<std::endl and found out it is always zero for every pair of coordinates). Another thing when I tried to print out output of get_texture coordinates(), like this:
`auto tex_coords = points.get_texture_coordinates();`
`std::cout << tex_coords[100].v << std::endl;`
It prints out zero again. So according to this, I guess - every point of the point cloud has color of point [0,0] in the RGB image. Maybe information, that I am able to display RGB frame without any problem could be helpful. 

This is the whole code: [https://github.com/eMrazSVK/JetsonSLAM/blob/master/pcl_testing.cpp](url)

Thank you for any suggestions!

ev-mp, 2018-06-25:
Hello @eMrazSVK ,
The sample you refer to does not use the API correctly - the order of API calls is reversed.
The pointcloud is generated with the `pc.calculate(..` call. If the other (color) image is not set at this stage, then the point cloud will be generated without texture coordinates.
It should be

    # Tell pointcloud object to map to this color frame
    pc.map_to(color)

    # Generate the pointcloud and texture mappings
    points = pc.calculate(depth)

In addition, it make sense to check that both color and depth frames are not empty.

See the a following [example](https://github.com/IntelRealSense/librealsense/blob/c8ee8fa1912b9297df13bfe097d527667fe0afba/wrappers/python/examples/export_ply_example.py) as a reference.
Additionally, review #1601 for PCL-specific info.

eMrazSVK, 2018-06-25:
Hello,

Thank you very much, yeah... You solved the problem very quickly, changing the order of the two lines did it. Now the colors are a bit weird, but I think this is solved, since the color was not the point of this question.

Many thanks!

closed 2018-06-25 by eMrazSVK