realsenseai/librealsense #1940: Can't build Dynamic Calibration API samples

jb455, 2018-06-22:
Hi,

I installed the latest calibration API from here: https://downloadcenter.intel.com/download/27415/Intel-RealSense-D400-Series-Dynamic-Calibration-Software-Tools (v2.5.2.0 at present), but when I try to open `CalibrationToolAPI\2.5.2.0\examples\DynamicCalibrator\RS400DynamicCalibrationSample.sln` in VS17 and build the solution it fails with 11 "cannot open include file" errors.

Is the source code available for DSDynamicCalibrationAPI.dll please? I'm currently trying to calibrate a camera pair using OpenCV/Matlab and finding the results from those differ from the Librealsense extrinsics so wanted to see how the LRS calibration works to see if I can transform my results to get it working.

I'm also wondering if it would be possible to modify the DynamicCalibrator app to obtain an extrinsic calibration between a RealSense camera and an external (rigidly connected) camera? I can see that the `ListCameras` function used in `Main.cpp` filters out non RS400 cameras; is there a hard limit on the cameras which can be used due to hardware constraints? I'm not asking you to do this for me; I'm happy to have a crack at it myself (though I may have questions!), just wanted an expert opinion on its plausibility before going to far in to it.

Thanks!

dorodnic, 2018-06-26:
Hi @jb455 
Could you try building it in Release instead of Debug?
Dynamic Calibration software is not open-source unfortunately. 
I need to check more about the external camera calibration

jb455, 2018-06-29:
Thanks, building in Release works though when I run it I get "dll missing" messages. Copying the relavent dlls from /bin to .../Build/x64-Release fixes and it runs without issue then.

I've solved my original problem myself somewhat. It turned out that my extrinsics were correct (after dividing the translation vector by 1000 to go from mm to m), but the `pointcloud.calculate()` call I was using to validate appeared to be using the wrong intrinsics: when the depth image is mapped to colour, the point cloud should be calcluated using the colour camera intrinsics instead of the depth camera (which I assume it was using). I extracted the point cloud generation code and called it manually with the colour extrinsics and obtained the correct results. I don't know if this problem is there all the time with the point cloud, or only when using the software device. 

It would still be useful to use the Intel calibration tool to obtain the extrinsics, but it's less important now I know I can use other calibration algorithms.

RealSense-Customer-Engineering, 2018-07-02:
[Realsense Customer Engineering Team Comment]
@jb455
Seems your problem is solved, can the ticket be close?

closed 2018-07-06 by RealSense-Customer-Engineering