realsenseai/librealsense #1941: python sample code for aligning point clouds of an object captured from different angles

bhomass, 2018-06-23:
Hi most of the sample code is in c++. I am working with python. Is there a python sample code available for aligning point clouds captured at different angles into one coherent point-cloud? 

Associated question (I am a newby in point cloud work) - In general how many capture angles does one need to create a good quality 3D view?

baptiste-mnh, 2018-06-25:
Hi!
You can check the Open3D library which offer a lot of functionalities for 3D reconstruction.
The lib is very well documented, even if you're a beginner it should be easy to deal with it :)
http://www.open3d.org/docs/tutorial/Basic/icp_registration.html

You should have a least ~75% of overlapping between two captures to get a good reconstruction.
The more you points of view, the best the reconstruction is.

dorodnic, 2018-06-25:
[opencv_contrib/rgbd](https://github.com/opencv/opencv_contrib/tree/master/modules/rgbd) also has some useful algorithms for point-cloud alignment.

bhomass, 2018-06-25:
@dorodnic would that be kinecfusion.? All the code is in c++ in that lib. Is there a simple rule for wrapping ?

dorodnic, 2018-06-26:
I personally tried [odometry_evaluation](https://github.com/opencv/opencv_contrib/blob/master/modules/rgbd/samples/odometry_evaluation.cpp) that accepts two RGBD images and calculates the transformation between them (using either ICP or other methods). 
I used C++ but you can most likely get the same functionality from python. Try `pip install opencv-contrib-python`, usually OpenCV APIs are translated very closely between C++ and python.

bhomass, 2018-06-26:
Thank you all for the answers.

@baptiste-mnh is there a ready api in open3d which can detect that the overlap is less than 75% and therefore suboptimal?

RealSense-Customer-Engineering, 2018-07-20:
[Realsense Customer Engineering Team Comment]
Hi @bhomass,

Are you able to have the implementation according to @baptiste-mnh's and @dorodnic's suggestions?

RealSense-Customer-Engineering, 2018-07-27:
[Realsense Customer Engineering Team Comment]
Hi @bhomass,

I will close this first. If any further question, you still can re-open or create another ticket.

closed 2018-07-27 by RealSense-Customer-Engineering