realsenseai/librealsense #1944: LabVIEW Wrapper HelloWorld Error

labview, 2018-06-24:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Firmware Version                   | 3.21.0.0 | 
| Operating System & Version | Win (10)| 
| Platform                                 | PC |
| SDK Version                            | 2.13 | 
| Language                            |  labview | 
| Segment			|  consumer   | 

### Issue Description
Goal: GreenScreen effect in Skype for Business using a new Razer Stargazer.
Status: No obvious Skype integration, 3rd party software failing, switched hats from consumer to hacker, error in Hello World.

Drivers look ok, RealSense Viewer 2.13.0 streams RGB and Coded-Light depth ok.

Running the "RealSense_HelloWorld_Depth_and_Color.vi" demo doesn't return data in either the image or intensity graph. Front panel Init errors about advanced mode look benign, no errors through config, start, capture 1, capture2 errors. "Close error" from the Front Panel:
Error calling rs2_stop (sensor:0000029BF2365E90):
 stop_streaming() failed. UVC device is not streaming!
Error calling rs2_close (sensor:0000029BF2365E90):
 close() failed. UVC device was not opened!
No Error
No Error

Highlight execution shows error cluster 1097 issues in the "rs3 start all 2.vi"  > "rs3 open multiple sensor.vi" back from a harmless looking DLL call. Might just need an DLL interface update.
Also saw a questionable Stream profile array input: two values in the array, the second value a ZERO. May indicate an upstream error I didn't probe.

<soapbox>Didn't dig past this point as I'm like 4 hours in on this off-topic :P Some new user feedback: the RealSense software stack, 3rd party app community, and documentation (even on Intel.com) is fragged, out of date, confusing and makes a high barrier to entry. Not sure how many consumers are still going at hour 4, and creating github accounts and getting involved in forums. Seems like quite a bit of frustration on forums for Razer, XSplit, and OBS.</soapbox>

agrunnet, 2018-06-24:
Sorry for your troubles. The Labview demo apps were made for the D4xx series of cameras. Not the SR300.

labview, 2018-06-24:
Thanks for the clarification.
Are any of the other wrappers made for SR300? Maybe I could port other examples to LabVIEW for completeness.

agrunnet, 2018-06-24:
That would be truly awesome. Yes, the wrapper should work, but you would need to change a few numbers, like resolution. Also, there are many of the options in LibRS that are mainly for D4xx, while some are for the SR300.

RealSense-Customer-Engineering, 2018-06-29:
[Realsense Customer Engineering Team Comment]
@labview
Thank you for the contribution in advance, I close this ticket first, if you have done the work, please file PR.

closed 2018-06-29 by RealSense-Customer-Engineering