realsenseai/librealsense #1945: Script bash files fail

Richard-Owen-3DI, 2018-06-25:
Hello, 

I hope someone can help me. I am having huge trouble getting my Intel RealSense R200 to be recognised by my computer.I am using Ubuntu 16.04 with kernel 4.13.0-45-generic

`lsusb` before and after connecting the RealSense to the computer (By USB 3.0) is identical and as follows:

```
Bus 002 Device 002: ID 8087:8000 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 004: ID 0930:0220 Toshiba Corp. 
Bus 001 Device 002: ID 8087:8008 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 004: ID 8086:0a80 Intel Corp. 
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 058f:6366 Alcor Micro Corp. Multi Flash Reader
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Also worth noting is that `sudo modprobe uvcvideo` produces no output. 

I have tried running the script `patch-uvcvideo-16.04.simple.sh ` found in /librealsense/scripts, the requisite packages install as expected, but then I get the following failure:

```
HEAD is now at 8a4cab0 UBUNTU: Ubuntu-4.4.0-128.154
./patch-uvcvideo-16.04.simple.sh: line 15: ../scripts/realsense-camera-formats_ubuntu-xenial.patch: No such file or directory
```

Any help would be greatly appreciated.

ev-mp, 2018-06-25:
Hello @Richard-Owen-3DI ,
The script you're running is obsolete and is not part of the installation instructions for either SDK v1 or v2.
Please use one of the official patches or DKMS (both patches and DKMS are backward-compatible with SDK v1.0 that you need):
1. [`patch-realsense-ubuntu-xenial.sh`for SDK v1.0](https://github.com/IntelRealSense/librealsense/blob/legacy/doc/installation.md#video4linux-backend-preparation)
2. [`patch-realsense-ubuntu-lts.sh` for SDK v2.0](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#prerequisites)
3. [DKMS package for Ubuntu 16](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#linux-distribution)

closed 2018-08-01 by RealSense-Customer-Engineering

AdamMiltonBarker, 2018-10-01:
Hi guys I am following legacy: https://github.com/IntelRealSense/librealsense/blob/legacy/doc/installation.md 

And this file does not exist, all of the links I am finding for the file are dead, do you know where to find the correct version, I am using R/F200 so using legacy library.