realsenseai/librealsense #1948: D435 "Import Error: DLL load failed: The Specified module could not be found"

jasonjsong, 2018-06-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.09.02.00 | 
| Operating System & Version |  Win 10  | 
| Platform                                 | PC|
| SDK Version                            | 2.0       | 
| Language                            |  python                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description

Hello all! I am trying to set up some code for my D435 Intel Realsense Camera, and have already installed the SDK as well as the GitHub librealsense folder. I have set the appropriate environmental variables (I believe), and after dragging the realsense dll file and the pyealsense2.pyd files into my project folder, as well as copying them into Python folder and setting the appropriate path variables, I am now met with the issue of "Import Error: DLL load failed: The Specified module could not be found". Any help would be greatly appreciated, thanks!

RealSense-Customer-Engineering, 2018-06-26:
[Realsense Customer Engineering Team Comment]
Hi @jasonjsong,

did you use the prebuild library from the Realsense SDK, rather than build it from source ?
if that is the case,
(1)Note that these binaries are built with Python 2.7, and cannot be import using Python 3)

(2) please check if either Miscrosoft Visual c++ 2015 Redistributable (X64) or Microsoft Visual c++2017 is installed on your PC, if no, please install either one
best
wwppyy

LeMajors, 2018-06-27:
I have still the same problems (https://github.com/IntelRealSense/librealsense/issues/1877 - unfortunately all hints doesn't work for me. Miscrosoft Visual c++ 2015 Redistributable (X64) is installed.)

You say, the binaries cannot import using Python 3 - @dorodnic said:
> Hi @LeMajors
Can you try the files installed with Intel.RealSense.SDK.exe? Under C:\Program Files\Intel RealSense SDK 2.0\bin we provide x86 and x64 versions for both python 2.7 and 3.6.

At .../Intel RealSense SDK 2.0/bin/ I can find only /x64 and /x86 folders - I assume these are for python 2.7 and _do not_ work for 3.6?

So there is no workaround and we have to build the library form source?

RealSense-Customer-Engineering, 2018-06-27:
[Realsense Customer Engineering Team Comment]
Hi @LeMajors,

at https://github.com/IntelRealSense/librealsense/tree/development/wrappers/python
the latest windows installation session  mentioned
"Windows users can install the RealSense SDK 2.0 from the release tab to get pre-compiled binaries of the wrapper, for both x86 and x64 architectures. (Note that these binaries are built with Python 2.7, and cannot be import using Python 3)"

by the way, on your setup, even with the VC++ redistributable (X64) is installed, using Python 2.7 still not able to import the pre-compiled library?
or you have to use python 3 for some reason?
if that the case, please see if there still  issues when build the library from source
best

LeMajors, 2018-06-27:
The D435 is part of a robot project that already has many thousands of lines of code. So I will not go back to 2.7.
As noob in terms of cmake I need some additional help.
What are the settings for cmake-gui (win10, 64Bit)?
- Where is the source code: "C:/.../RealSense D435/librealsense-2.12.0/src" doesn't work. What's the correct path?
- Where to build the binaries: I guess, it's up to me?
- Where and how can I select the option "BUILD_PYTHON_BINDINGS"?

Kind regards, Marcus

RealSense-Customer-Engineering, 2018-06-28:
[Realsense Customer Engineering Team Comment]
Hi @LeMajors
please find below for my inline comment
What are the settings for cmake-gui (win10, 64Bit)?
 - Where is the source code: "C:/.../RealSense D435/librealsense-2.12.0/src" doesn't work. What's the correct path?
[ww]
once you launch the cmake -gui
you should set the where the source code to C:/.../RealSense D435/librealsense-2.12.0/  (go one upper layer)
- Where to build the binaries: I guess, it's up to me?
yes, it is up to you,
ex :C:/.../RealSense D435/librealsense-2.12.0/build/, the CMAKE will create the folder for you if not exist

- Where and how can I select the option "BUILD_PYTHON_BINDINGS"?
once you set the directory right, the CMAKE will pop up the options list
then you need to check the BUILD_PYTHON_BINDINGS option

once checked, press configure button
and ensure the PYTHON EXCUTABLE is set to Python 3

ex :C:/Users/wangwill/AppData/Local/Programs/Python/Python36/python.exe

press generate button to generate the solution once completed you can press open project to launch the solution then build the library or sample code

under the build/release or build/debug folder, you should able to find realsense.dll and pyrealsense2.cp36-win_amd64.pyd which you need
you can copy both under C:\xxx\librealsensexxx\wrappers\python\examples\ and run the sample code to see if it work ex. python-tutorial-1-depth.py

LeMajors, 2018-06-28:
Hi @RealSense-Customer-Engineering 
thanks for your detailed informations.

Yes it's Win10, 64Bit, Intel Celeron CPU N3450.

When I set the two paths, there is no popup. Now I can just Add an Cache Entry (Name, Type, Value, Description) and Configure:
[image]
I tried all 3 choices "Visual Studio 15 2017", "Visual Studio 15 2017 Win64" , "Visual Studio 15 2017 ARM". All with this result:
**Error in configuration process, project files may be invalid**
The CMakeOutput.log says: "The system is: Windows - 10.0.17134 - AMD64" - but that's definitely not the case!?
What's wrong here?

Regards, Marcus

LeMajors, 2018-06-28:
Hi @RealSense-Customer-Engineering 
Yes, the files and folders are existing.
I also copied the folder to c:/ to shorten the path. To be sure I gave "everybody" full access to all files and folders. But It's the same result.
[image]

RealSense-Customer-Engineering, 2018-06-28:
[Realsense Customer Engineering Team Comment]
Hi @LeMajors
please refer to below link for the Windows librealsense installation guide which has pictures which is easier to follow
https://github.com/IntelRealSense/librealsense/blob/development/doc/installation_windows.md

LeMajors, 2018-06-28:
Ok, I came a little further...
After not seeing any options as in the screenshot, I reinstalled CMake.
Also I checked the properties of "CMakeLists.txt" - Windows said, it may be blocked because it comes from a different computer... I switched to "allow".
I guess, this was the main problem. Now he finds CMakeLists.txt because of this Error:

Generator
Visual Studio 15 2017 Win64 could not find any instance of Visual Studio.

So I reinstalled, but with no success. What I have installed is "Microsoft Visual C++ 2017 Redistributable (x64) - 14.14.26429".
Do I have to install something different?

RealSense-Customer-Engineering, 2018-06-29:
[Realsense Customer Engineering Team Comment]
Hi @LeMajors
it looks something wrong about your VC installation
please find below link which has similar issue as you

https://github.com/Valloric/YouCompleteMe/issues/2945

The `VC` folder should be there. Are you sure you selected `Desktop development with C++` in the `Workloads`tab during installation of Visual Studio 2017? Run again the `vs_Community.exe` installer, click `Continue`, wait for the products screen to appear, click on `Modify` under `Visual Studio Community 2017`, and check if `Desktop development with C++` is selected under `Windows`. If not, select it then click on the `Modify` button to install it. Once it's done, leave the installer and run again the `install.py` script.

> and do you mean i need to install vs2015 or vs 2013?

No, it should work fine with VS 2017.

 [image]

### **weikent** commented on Mar 12

It worked. Thank you for your help.
It is my fault, I don't select Desktop development with C++

LeMajors, 2018-06-30:
@RealSense-Customer-Engineering - Thank you for the help and your patience!

To summarize the problems:
- I had to allow access to the file "CMakeLists.txt" because it came from another computer.
- The "Visual Studio Community 2017" has to be installed. (I assumed that the installation would range from "Microsoft Visual C ++ 2017 Redistributable (x64) - 14.14.26429").

So for me everything is fine now. If it's also working for @jasonjsong  we can now close this issue.

jasonjsong, 2018-07-05:
Working for me as well now!

My primary issue was installing Visual Studio Community 2017, thanks for all the help everyone!

closed 2018-07-05 by jasonjsong